MarlinFirmware/Marlin #12272: Multiple Z stepper new feature - docs missing?

RudyBenner, 2018-10-30:
Seem to be a tad short on gcode for this. How do I try it?

// Automatic alignment of Z steppers for multi-Z-stepper and a probe
//#define Z_STEPPER_AUTO_ALIGN
#if ENABLED(Z_STEPPER_AUTO_ALIGN)
  // Define probe X and Y positions for Z1, Z2 [, Z3]
  #define Z_STEPPER_ALIGN_X { 10, 150, 290 }
  #define Z_STEPPER_ALIGN_Y { 290, 10, 290 }
  // Set number of iterations to align
  #define Z_STEPPER_ALIGN_ITERATIONS 3
  // Enable to restore leveling setup after operation
  #define RESTORE_LEVELING_AFTER_G34
  // Use the amplification factor to de-/increase correction step.
  // In case the stepper (spindle) position is further out than the test point
  // Use a value > 1. NOTE: This may cause instability
  #define Z_STEPPER_ALIGN_AMP 1.0
  // Stop criterion. If the accuracy is better than this stop iterating early
  #define Z_STEPPER_ALIGN_ACC 0.02
#endif

RudyBenner, 2018-10-30:
Sheepishly admits that the Gcode M34 is the secret sauce. I should have looked harder.

closed 2018-10-30 by RudyBenner

boelle, 2018-11-09:
i cant find that secret sauce, or maybe i'm just stupid

http://marlinfw.org/ does not have it yet. M34 is sd card sorting http://marlinfw.org/docs/gcode/M034.html

boelle, 2018-11-09:
#define Z_STEPPER_ALIGN_X { 36, 240}
#define Z_STEPPER_ALIGN_Y { 104.9, 104.9 }

does that look right?

RudyBenner, 2018-11-09:
This is my setup. You are only moving on one axis.

// Automatic alignment of Z steppers for multi-Z-stepper and a probe
#define Z_STEPPER_AUTO_ALIGN
#if ENABLED(Z_STEPPER_AUTO_ALIGN)
  // Define probe X and Y positions for Z1, Z2 [, Z3]
  #define Z_STEPPER_ALIGN_X { 10, 190 }
  #define Z_STEPPER_ALIGN_Y { 110, 110 }
  // Set number of iterations to align
  #define Z_STEPPER_ALIGN_ITERATIONS 6
  // Enable to restore leveling setup after operation
  #define RESTORE_LEVELING_AFTER_G34
  // Use the amplification factor to de-/increase correction step.
  // In case the stepper (spindle) position is further out than the test point
  // Use a value > 1. NOTE: This may cause instability
  #define Z_STEPPER_ALIGN_AMP 1.0
  // Stop criterion. If the accuracy is better than this stop iterating early
  #define Z_STEPPER_ALIGN_ACC 0.02
#endif

boelle, 2018-11-09:
yes... i have the mk42 bed so i need the probe to hit 2 points on left and right side of X and in middle of Y

that is why i dont move Y axis

boelle, 2018-11-10:
@RudyBenner quick Q

i got it working in first go, but now i'm wondering if it makes sense to run a G34 followed by G29 ?

i'm in doubt since the G34 ends with what looks like homing