MarlinFirmware/Marlin #12274: [bugfix 2.0.x]  No values for X,Y&Z on full graphic LCD

Bergerac56, 2018-10-30:
With the last bugfix 553622835911b520c6ba43bc03e71868f3972cb3 ( mks_sbase and full graphic LCD), there are no values appearing for X Y and Z on the info screen anymore. (And no blinking question marks when printer not yet "homed")

It seems to have been introduced in the long chain of changes made for the refactor of lcd menu (2bf18950ab2070e842c91057d5812dcaa1bff9a2).

mr-miky, 2018-10-30:
I had noticed it too yesterday but then I forgot to report it

gloomyandy, 2018-10-30:
Yep I'm seeing the same.

gloomyandy, 2018-11-01:
@thinkyhead do you have a handle on this, or is it worth me spending some time trying to work out what the problem is?

mr-miky, 2018-11-03:
Maybe this is the solution to the problem?
Check **first_page** instead of **page.page**

```
  // At the first page, regenerate the XYZ strings
    if (first_page) {
//  if (page.page == 0) {
    strcpy(xstring, ftostr4sign(LOGICAL_X_POSITION(current_position[X_AXIS])));
    strcpy(ystring, ftostr4sign(LOGICAL_Y_POSITION(current_position[Y_AXIS])));
    strcpy(zstring, ftostr52sp(LOGICAL_Z_POSITION(current_position[Z_AXIS])));
    #if ENABLED(FILAMENT_LCD_DISPLAY)
      strcpy(wstring, ftostr12ns(filament_width_meas));
      strcpy(mstring, itostr3(100.0 * (
          parser.volumetric_enabled
            ? planner.volumetric_area_nominal / planner.volumetric_multiplier[FILAMENT_SENSOR_EXTRUDER_NUM]
            : planner.volumetric_multiplier[FILAMENT_SENSOR_EXTRUDER_NUM]
        )
      ));
    #endif
  }
```

thinkyhead, 2018-11-03:
I just uploaded `bugfix-2.0.x` to my RAMPS board with graphical display and it's working fine for me.

closed 2018-11-03 by thinkyhead

gloomyandy, 2018-11-03:
Yep all working for me now. Thanks.

Bergerac56, 2018-11-05:
Works for me too