MarlinFirmware/Marlin #12277: Level Bed Corners compile fails after new menu refactor.

forkoz, 2018-10-30:
### Description

Linking .pioenvs\LPC1768\firmware.elf
.pioenvs/LPC1768/src/src/lcd/menu/menu_motion.cpp.o: In function `menu_motion()':
I:\3dprint\sBase\Marlin/Marlin\src\lcd\menu/menu_motion.cpp:482: undefined reference to `_lcd_level_bed_corners()'

When #define LEVEL_BED_CORNERS is defined marlin doesn't compile. I'm not sure if its confined to sBase or if other platforms share the same issue.

sl1pkn07, 2018-10-31:
same issue with sanguino board

this issue is commented in here https://github.com/MarlinFirmware/Marlin/commit/a489c9fa52261cd5836771f0f0c5f3dff20f696d#commitcomment-31094384

greetings

swilkens, 2018-10-31:
This error goes away when you define `LCD_BED_LEVELING` in `Configuration.h`

sl1pkn07, 2018-10-31:
^ this id for this:

``` 
  #if ENABLED(LEVEL_BED_CORNERS) && DISABLED(LCD_BED_LEVELING)
    MENU_ITEM(function, MSG_LEVEL_CORNERS, _lcd_level_bed_corners);
  #endif
```

sl1pkn07, 2018-10-31:
this is ok?

~~~patch
diff --git a/Marlin/src/lcd/menu/menu_motion.cpp b/Marlin/src/lcd/menu/menu_motion.cpp
index 4743b60ec..fe6a10d05 100644
--- a/Marlin/src/lcd/menu/menu_motion.cpp
+++ b/Marlin/src/lcd/menu/menu_motion.cpp
@@ -492,6 +492,7 @@ void menu_motion() {
   #endif
 
   #if ENABLED(LEVEL_BED_CORNERS) && DISABLED(LCD_BED_LEVELING)
+    #include "menu_bed_leveling.cpp"
     MENU_ITEM(function, MSG_LEVEL_CORNERS, _lcd_level_bed_corners);
   #endif
 
~~~

swilkens, 2018-10-31:
> this is ok?
> 
> ```diff
> diff --git a/Marlin/src/lcd/menu/menu_motion.cpp b/Marlin/src/lcd/menu/menu_motion.cpp
> index 4743b60ec..fe6a10d05 100644
> --- a/Marlin/src/lcd/menu/menu_motion.cpp
> +++ b/Marlin/src/lcd/menu/menu_motion.cpp
> @@ -492,6 +492,7 @@ void menu_motion() {
>    #endif
>  
>    #if ENABLED(LEVEL_BED_CORNERS) && DISABLED(LCD_BED_LEVELING)
> +    #include "menu_bed_leveling.cpp"
>      MENU_ITEM(function, MSG_LEVEL_CORNERS, _lcd_level_bed_corners);
>    #endif
>  
> ```

Sadly, no, now fails on `menu.cpp`:

> sketch\src\lcd\menu/menu.cpp:112: undefined reference to `_lcd_level_bed_corners()'

sl1pkn07, 2018-10-31:
yea. i commit gigatic mistake. can't include a cpp file. only headers (.h)

my second attemp:

~~~patch
diff --git a/Marlin/src/lcd/menu/menu_bed_leveling.cpp b/Marlin/src/lcd/menu/menu_bed_leveling.cpp
index 5c172b6cf..307b0ef98 100644
--- a/Marlin/src/lcd/menu/menu_bed_leveling.cpp
+++ b/Marlin/src/lcd/menu/menu_bed_leveling.cpp
@@ -286,6 +286,11 @@ void menu_bed_leveling() {
 
   const bool is_homed = all_axes_known();
 
+
+    #if ENABLED(LEVEL_BED_CORNERS) && DISABLED(LCD_BED_LEVELING)
+      MENU_ITEM(function, MSG_LEVEL_CORNERS, _lcd_level_bed_corners);
+    #endif
+
   // Auto Home if not using manual probing
   #if DISABLED(PROBE_MANUALLY) && DISABLED(MESH_BED_LEVELING)
     if (!is_homed) MENU_ITEM(gcode, MSG_AUTO_HOME, PSTR("G28"));
diff --git a/Marlin/src/lcd/menu/menu_motion.cpp b/Marlin/src/lcd/menu/menu_motion.cpp
index 4743b60ec..6fe41bfd3 100644
--- a/Marlin/src/lcd/menu/menu_motion.cpp
+++ b/Marlin/src/lcd/menu/menu_motion.cpp
@@ -491,10 +491,6 @@ void menu_motion() {
 
   #endif
 
-  #if ENABLED(LEVEL_BED_CORNERS) && DISABLED(LCD_BED_LEVELING)
-    MENU_ITEM(function, MSG_LEVEL_CORNERS, _lcd_level_bed_corners);
-  #endif
-
   //
   // Disable Steppers
   //
~~~

move the block to menu_bed_leveling.cpp

swilkens, 2018-10-31:
While it compiles, the "level bed" function in the LCD menu now does a homing sequence and nothing else.

forkoz, 2018-10-31:
Can't use  LCD bed leveling since I use auto bed leveling too. The level corners is to even out the bed before running UBL/Bilinear/etc. I tried to include some of the LCD bed leveling headers but it still fails. Don't have enough to see what's going on. Including a cpp file is not going to work :(

sl1pkn07, 2018-10-31:
build sucess with the seccond attemp. but if any dev can chek if is paste in the rigth zone.....

swilkens, 2018-10-31:
> build sucess with the seccond attemp. but if any dev can chek if is paste in the rigth zone.....

It builds, yes, but the corner leveling doesn't work. Calling the corner leveling function from LCD executes a homing sequence and nothing else.

sl1pkn07, 2018-10-31:
@thinkyhead ping?

closed 2018-10-31 by thinkyhead