MarlinFirmware/Marlin #12278: [bugfix-2.0.x] MKS SBASE halt on E movement after M164

Gav-in, 2018-10-31:
### Description

On the MKS SBASE v1.3, any attempt to move a virtual extruder after setting the mixing ratio with M164 makes the firmware/lcd unresponsive.
### Steps to Reproduce

1. Configure bugfix-2.0.x for BOARD_MKS_SBASE, and #define MIXING_EXTRUDER with any number of physical and virtual extruders.
2. Boot the firmware, and execute g-code that prints using all virtual extruders.
3. Add the following to the start of the g-code, and run it again.
     M163 S0 P1.00; Cyan factor
     M163 S1 P0.00; Magenta factor
     M163 S2 P0.00; Yellow factor
     M163 S3 P0.00; Black factor
     M163 S4 P0.00; White factor
     M164 S6; Save mixture to tool 6

**Expected behavior:** [What you expect to happen]
The M164 in the above example should replace the default mixture for virtual tool 6 with a user-defined mixture (which is actually the same color), so the print should be identical with or without the changes.

**Actual behavior:** [What actually happens]
The program proceeds normally until the user-modified virtual tool is issued a move instruction, at which point the system becomes unresponsive.

#### Additional Information
M163/M164/M165 were all working with 5 physical/256 virtual extruders on the SBASE and bugfix-2.0.x until a couple of weeks ago, when RETRACT_SYNC_MIXING was added, along with shared hardware spi for the lcd/sd.  There are many compiler warnings.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2532264/Config.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

AnHardt, 2018-10-31:
Before the `M164 S6` `T6` prints normal? (Just to decrease the search radius.)

Please add `#define MIXER_NORMALIZER_DEBUG` in `Configuration.h`. Will this give any useful hints? Please paste the output of
```
```

AnHardt, 2018-10-31:
> 5 physical/256 virtual extruders

V-tools are limited to 1-255. If `RETRACT_SYNC_MIXING` is defined 1-254. The 'retract tool' should not be selectable by T or M164. (tool number is a uint8_t. 256 will not fit in)

AnHardt, 2018-11-01:
Please try #12293

Gav-in, 2018-11-02:
Thank you so much!  I thoroughly tested all mixing functions, and everything works as described (except 254 virtual tools fails the check, lol.) I greatly appreciate the swift resolution.

closed 2018-11-05 by thinkyhead