MarlinFirmware/Marlin #12281: [2.0.x] M906.cpp:40:17: warning: unused variable 'index' [-Wunused-variable]

dahuafschmied, 2018-10-31:
### Description

i am using TMC2208 in UART mode just for E0.
X/Y/Z/Z2 are in standalone mode.

this results in an unused variable "index" in M906.

### Steps to Reproduce

`#define X_DRIVER_TYPE  TMC2208_STANDALONE
#define Y_DRIVER_TYPE  TMC2208_STANDALONE
#define Z_DRIVER_TYPE  TMC2208_STANDALONE
#define Z2_DRIVER_TYPE TMC2208_STANDALONE
#define E0_DRIVER_TYPE TMC2208`

compile.

#### Additional Information

my repository:
https://github.com/dahuafschmied/Marlin-for-Anycubic-i3-Mega/tree/bugfix-2.0.x

Ludy87, 2018-10-31:
try this

`\src\gcode\feature\trinamic`

[M906.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/2536295/M906.cpp.txt)

remove .txt

dahuafschmied, 2018-10-31:
this fixes the warning

dahuafschmied, 2018-10-31:
if you are looking at it already. there is an other one:
M911-M915.cpp:100:18: warning: unused variable 'hasNone' [-Wunused-variable]

Ludy87, 2018-10-31:
Defined RX/TX pins!

thinkyhead, 2018-10-31:
Also adding a sanity-check for cases where `MONITOR_DRIVER_STATUS` is enabled but there are no enabled TMC steppers supporting it.

closed 2018-10-31 by thinkyhead

teemuatlut, 2018-11-01:
`MONITOR_DRIVER_STATUS` can check all the supported TMC drivers and can only be enabled by using drivers with communication lines.
No sanity check needed (?).

thinkyhead, 2018-11-02:
If someone enables it in error, this added code ought to catch it…
```cpp
#if ENABLED(MONITOR_DRIVER_STATUS)

  #define M91x_USE(ST) (AXIS_DRIVER_TYPE(ST, TMC2130) || (AXIS_DRIVER_TYPE(ST, TMC2208) && PIN_EXISTS(ST##_SERIAL_RX)))
  #define M91x_USE_E(N) (E_STEPPERS > N && M91x_USE(E##N))

  #define M91x_SOME_X (M91x_USE(X) || M91x_USE(X2))
  #define M91x_SOME_Y (M91x_USE(Y) || M91x_USE(Y2))
  #define M91x_SOME_Z (M91x_USE(Z) || M91x_USE(Z2) || M91x_USE(Z3))
  #define M91x_SOME_E (M91x_USE_E(0) || M91x_USE_E(1) || M91x_USE_E(2) || M91x_USE_E(3) || M91x_USE_E(4) || M91x_USE_E(5))

  #if !M91x_SOME_X && !M91x_SOME_Y && !M91x_SOME_Z && !M91x_SOME_E
    #error "MONITOR_DRIVER_STATUS requires at least one TMC2130 or serial-connected TMC2208."
  #endif

  . . .

#endif
```