MarlinFirmware/Marlin #12282: [2.0.x] STM32 FastIO breaks PIN/PWM handling

hasenbanck, 2018-10-31:
#12276 seem to have introduce a regression. I tried to run my F7 board with the changes and it seems that the PIN handling is all over the place:
- Display is not working anymore (RepRapDiscount Full Graphic Controller)
- PWM pins are always on
- PWM configuration will trigger a HALT of the CPU (when calling pwm_start())

When I reverted the changes manually (FastIO), the board worked again as expected. I will try to debug the problem further.

hasenbanck, 2018-10-31:
I think the problems lies in a change from @thinkyhead. 

@ktand, you defined:
```c
#define _WRITE(IO, V) (FastIOPortMap[STM_PORT(digitalPin[IO])]->BSRR = (1UL << STM_PIN(digitalPin[IO]) << (V ? 0UL : 16UL)))
```

@thinkyhead changed it to:
```c
#define _WRITE(IO, V) (FastIOPortMap[STM_PORT(digitalPin[IO])]->BSRR = _BV32(STM_PIN(digitalPin[IO] + (V ? 0 : 16))))
```

When I use the original change of @ktand, the general PIN handling works again. The LCD is gibberish, but I think those are the delays that need some adjustment.

ktand, 2018-10-31:
It looks like there is a parenthesis at the wrong place:

#define _WRITE(IO, V) (FastIOPortMap[STM_PORT(digitalPin[IO])]->BSRR =
_BV32(STM_PIN(digitalPin[IO] + (V ? 0 : 16))))

I think it should be:

#define _WRITE(IO, V) (FastIOPortMap[STM_PORT(digitalPin[IO])]->BSRR =
_BV32(STM_PIN(digitalPin[IO]) + (V ? 0 : 16)))

hasenbanck, 2018-10-31:
@ktand , Yes you are right. Than change would fix it (just tried it on my board).

On a STM32F765VI @216 Mhz (activated I/D cache) I had to use following delays to make the display working again with no glitches:
```c
  #define CPU_ST7920_DELAY_1 DELAY_NS(1000)
  #define CPU_ST7920_DELAY_2 DELAY_NS(80)
  #define CPU_ST7920_DELAY_3 DELAY_NS(1000)
```
I think we should now include special cases for STM32F4 / F7 etc.
Thinking about it ... can't we use somehow the F_CPU value here to adjust the delay accordingly?

On another node, but since I happen to an test the change right now and you did the work on it:
The timer override by setting ```STEP_TIMER``` in the pin file doesn't seem to work as expected, since it seems that he uses still the value defined in the HAL_timers_STM32.h.

AnHardt, 2018-10-31:
1µs is huge.
Better try schemes where: DELAY_1 + DELAY_2 = DELAY_3 and DELAY_2 is as large as possible (DELAY_1 = 0)
That should give the data as much time as possible to stabilize before the clock comes.
Each of this delays is used at least 64\*128\*2 = 16,384 times a second wasting now 2080*16,384 = 34ms/s. It's worth to optimize this.

If you really need to set `DELAY_NS(1000)` this leads to the question if the `DELAY_NS()` macro on your processor is lasting the expected time or is about 3 times to fast.

hasenbanck, 2018-10-31:
DELAY_NS() is using the ARM specific functions of HAL/shared/Delay.h. I will try to test these functions by using a GPIO pin and an oscilloscope when I have the time to do so.

I will also try to optimize the timings as explained by you.

hasenbanck, 2018-10-31:
Code:
```c
void loop() {
  digitalWrite(PE5, HIGH);
  DELAY_NS(500000);
  digitalWrite(PE5, LOW);
  DELAY_NS(500000);
}
```

Waveform:
[image]

I would have expect to have a period of ~1000us and not ~440us.

AnHardt, 2018-10-31:
From https://blueprints.launchpad.net/gcc-arm-embedded/+spec/delay-cycles

> Also something to consider is that more feature rich Cortex' may simply ignore NOPs. They enter the CPU pipeline then, but get discarded before they consume time. This info is picked up from one of the Cortex-M user manuals.

ktand, 2018-10-31:
> Yeah, most likely we need to use another method for Cortex-M7 MCU. I think
> starting from Cortex-M4 there is a cycle count register (CYCCNT) that we
> could use for that purpose.
>
> I think I will try to write a solution in the next couple days.
>
> @ktand <https://github.com/ktand> I think we can leave the delay values
> as they are right now, until we implemented a better delay function for the
> M7 (and maybe also M4) processors.

Using the CYCCNT for delays on F4/F7 sounds like an interersting idea!

AnHardt, 2018-10-31:
Maybe http://www.anthonyvh.com/2017/05/18/cortex_m-cycle_counter/ is helpful.

hasenbanck, 2018-10-31:
> Maybe http://www.anthonyvh.com/2017/05/18/cortex_m-cycle_counter/ is helpful.

Using that code example I get a nice 1ms delay using the above blink loop. I think it should be nice if we activate this method of delaying for all compatible platforms (currently cortex-m3, m4 and m7). I would like to extend the Delay.h for this. I think we also need a place to store the "enableCycleCounter()" somewhere. I don't think we should include that setup code in the HALs, since we would have duplicated code in all Cortex M3+ HALs then. Maybe we could include the "enableCycleCounter()" somwhere in the Marlin.cpp?

p3p, 2018-10-31:
> Maybe we could include the "enableCycleCounter()" somwhere in the Marlin.cpp?

It's a platform dependant implementation detail of delay so it has to be behind the HAL, as usual there may be some code repetition on platforms that share an architecture but not much else (CMSIS version ect), the shared section is supposed to mitigate this if its possible.

hasenbanck, 2018-10-31:
> It's a platform dependant implementation detail of delay so it has to be behind the HAL, as usual there may be some code repetition on platforms that share an architecture but not much else (CMSIS [...]

Since we are speaking of a CMSIS feature that would be used by all affected HALs, the current Delay.h seems like the right place. We could add the "enableCycleCounter()" in every affected HAL_init() function, that maybe the most clean solution.

AnHardt, 2018-10-31:
```cpp
inline volatile uint32_t getCycleCount () { return DWT->CYCCNT; }

inline void delayCycles (uint32 const numCycles) {
  uint32_t const startCycles = getCycleCount();
  while ((getCycleCount() - startCycles) < numCycles) { }
}
```
does not look very overflow save. Delay will be to short then. Maybe set CYCCNT to 0. How often will the overflow occur? Does it make sense to call `delayCycles ()` multiple times? Called with a constant there could be a overflow warning.
Precalculating the 'stop' will increase the resolution a bit.

hasenbanck, 2018-10-31:
Okay guys, I will open a PR and you can then add your comments / improvements there.

ktand, 2018-10-31:
Because of the overflow issue, would it be better to just use CYCCNT to
calibrate a delay loop at startup (with interrupts disabled)?

AnHardt, 2018-10-31:
Resetting CYCCNT to 0 is not an option. Will be used, at least, for the display SPI delays and in the stepper interrupt.

AnHardt, 2018-10-31:
Calling 
```
t = T/n
for (int i = 0; i<n; i++)
  delayCycles(t); 
```
would reduce the possible error due to overflows to T/n

hasenbanck, 2018-10-31:
Opened a PR. Please feel free to add code changes and suggestions.

hasenbanck, 2018-11-02:
Resolved by #12283 and #12284

closed 2018-11-02 by hasenbanck