MarlinFirmware/Marlin #12285: Marlin 2.0 - Sanguino - Zonestar LCD with ADC keypad doesn't compile

towe96, 2018-10-31:
### Description

When trying to compile for a Tronxy XY-100, it fails to compile because of what seems to be a mixup between ADC keypad and RepRapWorld keypad code.

### Steps to Reproduce

1. Board is "BOARD_MELZI"
    LCD is "ZONESTAR_LCD"

2. Trying to compile

**Expected behavior:** Compiles, buttons / printer works

**Actual behavior:** Fails to compile

#### Additional Information

The errors are about "EN_REPRAPWORLD_KEYPAD_xxx" not being defined, with xxx being each of the RepPapWorld Keypad's buttons. Also "REPRAPWORLD_KEYPAD_MOVE_MENU" throws an error for being undefined.
I attempted modifying the "ultralcd.cpp" and "ultralcd.h" files to be similar to what they were in 1.1.8, and got it to compile, but then the buttons didn't work at all.
It also doesn't compile with the configuration found in the "examples" folder for the XY-100.
In order to even see the relevant error codes, I removed the pin-redefinitions in the "SANGUINOLOLU_11.h" file by commenting out lines 211 through 216 and 220 to 221, but those were only warning that were spammed in the Arduino IDE's compile window.
The problem is 100% reproducable with the stock example configuration (after preventing the warning lines).

Ludy87, 2018-10-31:
try this file

[ultralcd.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/2535779/ultralcd.cpp.txt)

towe96, 2018-10-31:
Dear Ludy87, your file has fixed the issue I've had, many thanks. Guess it just needs some tidying up now.
It was roughly equivalent to my approach of deleting RepRapWorld-lines until it compiled, guess I must've missed something.

Ludy87, 2018-11-01:
The new Bugfix is online

jalanjarosz, 2018-11-02:
> 
> 
> try this file
> 
> `\src\lcd`
> 
> [ultralcd.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/2535779/ultralcd.cpp.txt)
> 
> `\src\lcd\menu`
> 
> [menu_bed_leveling.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/2535805/menu_bed_leveling.cpp.txt)
> 
> remove ".txt"

With these files, for an Anet A8 board, I received the following error when compiling the base config files:

Arduino: 1.8.7 (Windows 10), Board: "Anet V1.0"

sketch\src\lcd\ultralcd.cpp: In function 'void lcd_update()':

sketch\src\lcd\ultralcd.cpp:680:26: error: 'LCD_CLICKED' was not declared in this scope

     if (UBL_CONDITION && LCD_CLICKED) {

                          ^

exit status 1
Error compiling for board Anet V1.0.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jalanjarosz, 2018-11-02:
@Ludy87 
After adding two lines, commenting out 2 lines, and adjusting the last line, I was able to compile keeping in #define ZONESTAR_LCD. 

I will add in ABL then FILRUNOUT to my sketch and advise if I see any other issues.  

Thanks!

closed 2018-11-03 by thinkyhead