MarlinFirmware/Marlin #12287: Corexy Axis homing correctly but homing to wrong direction when auto homing

Louis-Cypher, 2018-10-31:
### Description

After Upgrading to Marlin 2.0.x I checked direction of motion for all axes of my printer (corexy; sparkcube derivate) as well as endstops. The printer homes correctly in case x and y axis is homed separately. As soon as i do an atuohome or G28. The printer homes x correctly but y is homed to the wrong direction (opposite direction of direction uses when homing y separately.  

### Steps to Reproduce

1. Home to y 
2. G28
3. fix broken parts ;-)

**Expected behavior:** Home to direction of single axis homing
**Actual behavior:** Homes to opposite direction

#### Additional Information
- I have very coarse motions on x and y wheras z is running as expected.
- I converted my config using marlin-config
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2536598/configs.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Ludy87, 2018-10-31:
INVERT Y true

Louis-Cypher, 2018-10-31:
Thanks, a lot. This would change my y- axis direction as well, right? Why does this only affect when doing autohome but not homing y? I'm still confused

Ludy87, 2018-10-31:
execute G0 Y50 and see if all right

Louis-Cypher, 2018-10-31:
I already did that. Axis is moving to the right direction and endstop is in right direction. As I already pointed out homing of single axis (all of them!) works, autohome doesn't

Louis-Cypher, 2018-10-31:
Inverting Y messes up axis direction as expected

Louis-Cypher, 2018-10-31:
In my opinion homing of single axis has to happen in the same direction autohoming uses. Maybe I'm wrong...

Louis-Cypher, 2018-10-31:
I cleared EEPROM once more and the problem is gone

closed 2018-10-31 by Louis-Cypher