MarlinFirmware/Marlin #12291: How do I Set up Dual Y motors useing the Z output on Ramps 1.4

QuestionMKII, 2018-11-01:
I'm trying to get the following setup:
Output X - single motor that moves in the X direction
Output Y - single motor that moves in the Z direction
Output Z - two motors that move in the Y direction
Output E0 - the extruder
Output E1 - Not used

I'm very new to all of this. I have no idea what I'm doing. The Printer I'm using is a Type A Machines Series 1 (plywood). Please help.

gloomyandy, 2018-11-01:
The obvious question is why are you doing this? Why swap Y and Z over in this way? Why not simply change the plug arrangement. Also I think if you are going to get much help you need to provide details of what control board you are trying to use and possibly what stepper controllers (if they are not part of the board). You should probably also provide information on how the dual stepper setup is wired (is there a single plug to the control board, do you have two plugs), photos may help.

Finally this is not really an "issue" with Marlin. There are probably better ways to get support with how to set things up than asking here. You could try one of the many 3d printing forums, use the Marlin facebook group, or ask on the MArlin discord channels.

QuestionMKII, 2018-11-01:
> The obvious question is why are you doing this? Why swap Y and Z over in this way? Why not simply change the plug arrangement. Also I think if you are going to get much help you need to provide [...]

Thanks for the suggestions. As I stated I'm very new, I don't even know what 3d printing forums exist (aside from reddit). The printer is a "Type A Machines Series 1 (plywood)" running on a ramps 1.3 or 1.4. I can't identify the controllers (there white, that's all I know). The printer has 2 Y motors and one Z motor. currently both Y motors are plugged in to the 2 Z pin sets (this is how I received the printer). The printer itself is lazercut plywood construction. i don't fancy taking it apart to build a duel Z printer. I'll ask around elsewhere, but if you think of a solution, let me know.

Moyster, 2018-11-01:
If your board is a RAMPS board, you can refer to the online schematics and various wiring diagrams.  
You can also simply solder an adapter (just make sure to replicate your current setup, check if you have parallel or serial wired motors), so you can plug two motors in a single driver pins.
It would be the best thing to do, just pair the "axis motors" to their respective output, since you're (i guess) moving to marlin, you won't have to swap drivers roles to match your (custom) assembly.  
If you have a 5th stepper driver, you can check configuration_adv.h for "Z_DUAL_STEPPER_DRIVERS" option, and wire your second Z motor to E1, this way you can simply plug Z1 to Z and Z2 to E1, and keep Y on the Y stepper driver.

closed 2018-11-05 by thinkyhead