MarlinFirmware/Marlin #12292: Anycubic Kossel Linear Plus example config compilation problem

tig33r, 2018-11-01:
Stock Trigorilla board. Newest bugfix 2.0 with config from examples folder with changes below:

```
#define ANYCUBIC_KOSSEL_PLUS

// Anycubic Probe version 1 or 2 see README.md; 0 for no probe
#define ANYCUBIC_PROBE_VERSION 2

// Heated Bed:
// 0 ... no heated bed
// 1 ... aluminium heated bed with "BuildTak-like" sticker
// 2 ... ultrabase heated bed
#define ANYCUBIC_KOSSEL_ENABLE_BED 1
```

Error log:

```
Arduino:1.8.7 (Windows 10), Płytka:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src\lcd\menu\menu_bed_leveling.cpp:31:0:

sketch\src\lcd\menu\menu_bed_leveling.cpp: In function 'void menu_bed_leveling()':

sketch\src\lcd\menu\menu_bed_leveling.cpp:253:50: error: 'zprobe_zoffset' was not declared in this scope

     MENU_ITEM_EDIT(float52, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX);

                                                  ^

sketch\src\lcd\menu\menu.h:275:28: note: in definition of macro '_MENU_ITEM_PART_2'

       menu_action_ ## TYPE(__VA_ARGS__); \

                            ^

sketch\src\lcd\menu\menu.h:289:37: note: in expansion of macro 'MENU_ITEM_P'

 #define MENU_ITEM(TYPE, LABEL, ...) MENU_ITEM_P(TYPE, PSTR(LABEL), ## __VA_ARGS__)

                                     ^

sketch\src\lcd\menu\menu.h:315:42: note: in expansion of macro 'MENU_ITEM'

 #define MENU_ITEM_EDIT(TYPE, LABEL, ...) MENU_ITEM(_CAT(setting_edit_,TYPE), LABEL, PSTR(LABEL), ## __VA_ARGS__)

                                          ^

sketch\src\lcd\menu\menu_bed_leveling.cpp:253:5: note: in expansion of macro 'MENU_ITEM_EDIT'

     MENU_ITEM_EDIT(float52, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX);

     ^

sketch\src\lcd\menu\menu_bed_leveling.cpp:253:50: error: 'zprobe_zoffset' was not declared in this scope

     MENU_ITEM_EDIT(float52, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX);

                                                  ^

sketch\src\lcd\menu\menu.h:279:95: note: in definition of macro '_MENU_ITEM_PART_2'

       lcd_implementation_drawmenu_ ## TYPE(encoderLine == _thisItemNr, _lcdLineNr, PLABEL, ## __VA_ARGS__); \

                                                                                               ^

sketch\src\lcd\menu\menu.h:289:37: note: in expansion of macro 'MENU_ITEM_P'

 #define MENU_ITEM(TYPE, LABEL, ...) MENU_ITEM_P(TYPE, PSTR(LABEL), ## __VA_ARGS__)

                                     ^

sketch\src\lcd\menu\menu.h:315:42: note: in expansion of macro 'MENU_ITEM'

 #define MENU_ITEM_EDIT(TYPE, LABEL, ...) MENU_ITEM(_CAT(setting_edit_,TYPE), LABEL, PSTR(LABEL), ## __VA_ARGS__)

                                          ^

sketch\src\lcd\menu\menu_bed_leveling.cpp:253:5: note: in expansion of macro 'MENU_ITEM_EDIT'

     MENU_ITEM_EDIT(float52, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX);

     ^

exit status 1
```

Ludy87, 2018-11-01:
In the PR is a Fix https://github.com/MarlinFirmware/Marlin/pull/12288/commits/e42a9281e83e9dd7c599a83dc0010c9399b1bbed#diff-8911774161fce3416fac078d8bec98a7

tig33r, 2018-11-01:
Ok then I will wait to include this pr and close this issue.

closed 2018-11-01 by tig33r

Ludy87, 2018-11-01:
Try the new Bugfix

tig33r, 2018-11-02:
Already closed and compiling fine.