MarlinFirmware/Marlin #12295: [FR] Toggle endstop lines to OUTPUT and LOW when not in use for EMC/ESD compliance

marcio-ao, 2018-11-01:
As a commercial vendor of printers, we are constantly struggling to get our printers to pass FCC radiated emissions and ESD tests. Although we have typically hacked our FW to solve such problems ourselves, I would like to start a discussion about making such improvements upstream in a more general fashion.

The electronics box should be a Faraday cage, but in practice wires leading outside of allow electrical noise to leak. The major contributor to this are the endstop lines that are left as high-impedance INPUT lines when not in use. Inside the electronics box, stray high frequency noise from the electronics is coupled onto those wires; outside the electronics box, those wires act as an antenna and radiate those signals outwards.

The other problem has to do with ESD. When those lines are left ungrounded, they can develop a high voltage potential which can eventually reboot the printer. We are seeing this happen unintentionally due to friction from the moving parts of some of our printer, as well as during ESD testing where we are purposely subjecting the printer to external HV charges. Grounding unused lines seems to help with the situation by not allowing charges to accumulate as much.

While I understand that most DIY folks don't give a hoot about FCC certification, we suspect that EMC and ESD issues could make a home-built printers less reliable for the average user. I would like to start a discussion about this and what can be done to mitigate problems.

As a first course of action, I suggest setting all endstop and probe pins to OUTPUT LOW when they are not being actively used.

Thoughts?

gloomyandy, 2018-11-01:
What impact will doing this have on various different types of sensors, in particular those that may actively drive the input lines (like some inductive probes may do for instance)?

ejtagle, 2018-11-01:
@marcio-ao : I do understand your concerns (i have myself designed for EMC/ESD compliance), but the main problem here is that actively driving outputs to GND, even if i completely agree that is the recommended advice from the manufacturers of all microprocessors (unused pins should be set to input and connected to either VCC or GND, or they should be set to output and driven LOW)
But from the hobbist point of view, setting unused io pins to output is dangerous, as any short circuit or bad connection can destroy the microcontroller.

So, it is a hard decision... i would make it optional.

The other choice is a hardware mod. I have used an small RC network on any input (at the processor end!) to make sure ESD does not damage anything, and EMI is not radiated. An small 10n/100n SMD capacitor as close as possible to the microprocessor is more than enough

closed 2019-02-14 by marcio-ao