MarlinFirmware/Marlin #12297: Homing and probing.

magnusjjj, 2018-11-01:
### Description

When Z_SAFE_HOMING is enabled on my printer, it homes to the wrong position on the Y axis, but not the X axis.

### Steps to Reproduce

Unable to reproduce on other printers. Have reproduced on this printer with both stable 1.1.9, bugfix 1.1.x, and bugfix 2.0.x.

**Expected behavior:** [What you expect to happen]

The probe would move to the center of the board, and then probe.

**Actual behavior:** [What actually happens]

When homing with Z_SAFE_HOMING, the probe does not go to the center of the bed on the Y axis, but instead goes to the absolute limit of the bed. The X axis however does.
I have disabled auto leveling, and it changed nothing, so it cant be any probe coordinate things from there. I have had it enabled before troubleshooting, and it *also* gets offset the same way.

When moving with g0/g1 with the same speed settings as the homing speed, it moves to *exactly* the *correct* position. All of the steps/mm are also calibrated and measured. If I set Z_SAFE_HOMING_Y_POINT to 0 instead of the ((Y_BED_SIZE) / 2) it homes to the center of the board. The option to make the center 0,0 is not enabled.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlinconfig.zip](https://github.com/MarlinFirmware/Marlin/files/2539488/Marlinconfig.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute)) for additional guidelines.

GMagician, 2018-11-02:
Did you execute M502 and m500 after firmware update?

magnusjjj, 2018-11-02:
I did not, but I did now, thanks for the tip. Still homes incorrectly.

magnusjjj, 2018-11-02:
Actually, it says that the EEPROM is disabled.

GMagician, 2018-11-02:
Sure `Y_PROBE_OFFSET_FROM_EXTRUDER` value is correct?

Enable `DEBUG_LEVELING_FEATURE` , M111 S255 and finally G28. Then post report you get

magnusjjj, 2018-11-02:
#define X_PROBE_OFFSET_FROM_EXTRUDER 39  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -47  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]

That's not enough for the ~105 discrepency that's measured. The config is linked in the zip file above :). (Not being bitter or anything! I just figured it might be annoying or not even that visible link :) ). Can pastebin if you like instead.

I will try the debug_leveling_feature when I get home. Might be a bright idea :)

magnusjjj, 2018-11-02:
I am an absolute idiot. The offset to the probe was wrong on the Y axis. I had tried.. so many things over the last *three days*. And the guys over at #reprap on irc had no clue either, though they were very helpful.

Suggestion, then. The -front +behind is not that helpful, since on my printer this was the front. And I matched the Y axis direction to match the other printer I own.
I am not sure if marlin does something with the 'direction', but otherwise, would it make more sense to instead write something more.. precise? I mean, I should have figured it out based on that it should have been added to the position, being a programmer, but.. uh.
Is there anything in the manual or configuration that says what marlin considers 'front'?

Will supply patch if there are suggestions for additional text there. Am tired, ashamed+happy+angry at the same time :').

GMagician, 2018-11-02:
> The -front +behind is not that helpful, since on my printer this was the front. And I matched the Y axis direction to match the other printer I own.

In configuration.h, just above the parameters, there is a simple drawn that explain how to write value and sign. if home is down/left a positive value means probe is far from home than nozzle

closed 2018-11-03 by magnusjjj