MarlinFirmware/Marlin #12299: Bug 2 vs Anet A8

RudyBenner, 2018-11-01:
Tried to squeeze Bugfix 2.0 into an Anet 8 board. There now seem to be problems.
Compiler says nasty things. 
The files are from the example files in Anet/A8.

[Rudy.zip](https://github.com/MarlinFirmware/Marlin/files/2539525/Rudy.zip)

Ludy87, 2018-11-01:
Same Error https://github.com/MarlinFirmware/Marlin/issues/12285

Try the new Bugfix

RudyBenner, 2018-11-01:
same result

jalanjarosz, 2018-11-02:
I grabbed LCD.cpp files from error #12285 using Anet A8 sample config and received the following errors during compile:

C:\Users\jjarosz\AppData\Local\Temp\arduino_build_769502\sketch\src\lcd\ultralcd.cpp: In function 'void lcd_update()':

C:\Users\jjarosz\AppData\Local\Temp\arduino_build_769502\sketch\src\lcd\ultralcd.cpp:680:26: error: 'LCD_CLICKED' was not declared in this scope

     if (UBL_CONDITION && LCD_CLICKED) {

                          ^

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
Error compiling for board Anet V1.0.

thinkyhead, 2018-11-02:
> `if (UBL_CONDITION && LCD_CLICKED) {`

The current code is different. Try the very latest `bugfix-2.0.x` branch.

RudyBenner, 2018-11-02:
Compiling .pioenvs\melzi_optiboot\src\src\libs\buzzer.cpp.o
Compiling .pioenvs\melzi_optiboot\src\src\libs\hex_print_routines.cpp.o
In file included from Marlin\src\lcd\ultralcd.cpp:29:0:
Marlin\src\lcd\ultralcd.cpp: In function 'void lcd_update()':
Marlin\src\lcd\ultralcd.h:414:27: error: 'REPRAPWORLD_KEYPAD_MOVE_MENU' was not declared in this scope
#define LCD_CLICKED() REPRAPWORLD_KEYPAD_MOVE_MENU
^
Marlin\src\lcd\ultralcd.cpp:680:26: note: in expansion of macro 'LCD_CLICKED'
if (UBL_CONDITION && LCD_CLICKED()) {
^
*** [.pioenvs\melzi_optiboot\src\src\lcd\ultralcd.cpp.o] Error 1
 [ERROR] Took 134.62 seconds 

 [SUMMARY] 
Environment megaatmega2560      	[SKIP]
Environment megaatmega1280      	[SKIP]
Environment at90usb1286_cdc     	[SKIP]
Environment at90usb1286_dfu     	[SKIP]
Environment DUE                 	[SKIP]
Environment DUE_USB             	[SKIP]
Environment melzi_optiboot      	[ERROR]

RudyBenner, 2018-11-02:
Commenting out 
#define ZONESTAR_LCD 
fixes the problem of course, but then no display

jalanjarosz, 2018-11-02:
Just downloaded Marlin-bugfix-2.0.x-2.zip with an internal date of 11/1/2018 11:32 PM and received the following errors:

`In file included from C:\Users\jjarosz\AppData\Local\Temp\arduino_build_168880\sketch\src\lcd\ultralcd.cpp:29:0:

C:\Users\jjarosz\AppData\Local\Temp\arduino_build_168880\sketch\src\lcd\ultralcd.cpp: In function 'void lcd_update()':

C:\Users\jjarosz\AppData\Local\Temp\arduino_build_168880\sketch\src\lcd\ultralcd.h:414:27: error: 'REPRAPWORLD_KEYPAD_MOVE_MENU' was not declared in this scope

     #define LCD_CLICKED() REPRAPWORLD_KEYPAD_MOVE_MENU

                           ^

C:\Users\jjarosz\AppData\Local\Temp\arduino_build_168880\sketch\src\lcd\ultralcd.cpp:680:26: note: in expansion of macro 'LCD_CLICKED'

     if (UBL_CONDITION && LCD_CLICKED()) {

                          ^

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
Error compiling for board Anet V1.0.`

I too can get to compile when commenting out `#define ZONESTAR_LCD`

Ludy87, 2018-11-02:
Try this fix https://github.com/Ludy87/Marlin/commit/b31a9300b8a717d9b677b52ecb364b5d10b76dbc

Edit this file ans compile

jalanjarosz, 2018-11-02:
I have a clean compile now.  Thanks!
I will add ABL and FILRUNOUT now and test further.
Thanks!

RudyBenner, 2018-11-02:
> Try this fix [Ludy87@b31a930](https://github.com/Ludy87/Marlin/commit/b31a9300b8a717d9b677b52ecb364b5d10b76dbc)
> 
> Edit this file ans compile

Sorry, for some reason, I am not seeing what your fix is.

closed 2018-11-03 by thinkyhead

RudyBenner, 2018-11-03:
Got it. !!!! Just follow the bouncing ball.