MarlinFirmware/Marlin #12303: anet a8 error

brushmaker, 2018-11-02:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

brushmaker, 2018-11-02:
1.1.9 will not compile i got an error  pasting"/"SPIMaster in slave Outpin"/" and "DDR' doe not give a valid preprocessing token   Copy Error message
I do not know what this means and what to do.
[image]

closed 2018-11-02 by brushmaker

thinkyhead, 2018-11-02:
Looks like you figured it out: Use the newest Arduino IDE or use PlatformIO.