MarlinFirmware/Marlin #12304: [bugfix-2.0.x] BLTouch sometimes drops pin while probing

manutenfruits, 2018-11-02:
### Description

When running command G29 to level the bed, some BLTouch devices drop the pin after *some* probing points.

It is worth noting that this started happening after a recent firmware update, when I merged up to commit [5536228]([https://github.com/MarlinFirmware/Marlin/commit/553622835911b520c6ba43bc03e71868f3972cb3). [Firmware config here](https://github.com/manutenfruits/Marlin/commit/15969db43c5f19e94794de23245b7f3d582a833b)

I reverted back to some old firmware that I had ([this branch](https://github.com/manutenfruits/Marlin/tree/hypercube-32)) and G29 ran with **no issues**.

I'm sorry I haven't updated in forever, so the change might be harder to spot. Happy to try any fixes you guys suggest.

### Steps to Reproduce

1. Home all axes
2. Run G29 to level the bed

**Expected behavior:** For each probe point BLTouch drops, probes, stays stowed and moves to the next point.

**Actual behavior:** For *some* probe points, BLTouch drops, probes, which stows the probe, and while raising Z before moving to the next stop, the probe drops.

#### Additional Information

Demo here: https://photos.app.goo.gl/HzvxJPnnp3VuTfNu7

**THE DEMO VIDEO DOESN'T CORRESPOND TO EITHER OF THE FOLLOWING GRID TABLES**
Bed Grid on the reported firmware (with bug):
```
Bilinear Leveling Grid:
      0      1      2      3      4      5      6
 0 -1.389 -1.226 -1.127 -1.035 -0.942 -0.930 -0.977
 1 -0.777 -0.651 -0.519 -0.396 -0.329 -0.316 -0.414
 2 -0.470 -0.337 -0.175 -0.091 -0.044 -0.111 -0.229
 3 -0.330 -0.137 +0.015 +0.118 +0.180 +0.107 -0.051
 4 +0.067 +0.196 +0.304 +0.376 +0.374 +0.276 +0.109
 5 +0.339 +0.451 +0.538 +0.580 +0.562 +0.465 +0.308
 6 +0.346 +0.429 +0.496 +0.534 +0.498 +0.380 +0.258
```

Bed Grid on the old firmware (no bug):
```
Bilinear Leveling Grid:
      0      1      2      3      4      5      6
 0 -1.362 -1.172 -1.081 -0.966 -0.881 -0.841 -0.906
 1 -0.710 -0.590 -0.465 -0.325 -0.245 -0.240 -0.315
 2 -0.400 -0.255 -0.104 -0.019 +0.026 -0.029 -0.134
 3 -0.202 -0.017 +0.118 +0.238 +0.276 +0.226 +0.051
 4 +0.178 +0.317 +0.428 +0.498 +0.499 +0.404 +0.254
 5 +0.439 +0.559 +0.664 +0.714 +0.714 +0.629 +0.454
 6 +0.459 +0.495 +0.570 +0.590 +0.535 +0.440 +0.295
```

landodragon141, 2018-11-02:
Try tightening the screw on the top slightly, like a quarter turn.

manutenfruits, 2018-11-02:
@Stephane-80 this is solely a firmware issue, as reverting the firmware fixes it. Also it's worth noting again that leveling works fine either way. The board is an MKS Sbase too, with bugfix-2.0.x, and the BLTouch is an original one.

@thinkyhead I can try to tweak `Z_CLEARANCE_BETWEEN_PROBES` but that sounds like a workaround, not a fix, right? Also, I am running `bugfix-2.0.x`, as mentioned on my first comment. I will keep an eye on new updates either way.

p3p, 2018-11-02:
It could be an issue with the new lpc176x frameworks servo implementation, I didn't test with BLTouch devices as I don't have one but others did and didn't report anything. Would have been introduced after https://github.com/MarlinFirmware/Marlin/commit/213e94bce2fd74b10014557bd6801b45193a08f8 in this [commit](https://github.com/MarlinFirmware/Marlin/commit/3d5be2ea4bac370e05b8acc61a9fba2684e5131e) if that is the case,

Stephane-80, 2018-11-02:
@manutenfruits ok thanks , yes, before the updates of this summer I had no problems indeed, the problem appeared then.

@dough29 antclabs BLTouch here .

Stephane-80, 2018-11-02:
I just saw something strange
this time I had the problem after the last point of bed leveling (G29) only
the pine was out 
I wanted to go back by pushing it with my finger but it came out every time
and at that moment came to me the idea of making "M280 P0 S90" and he came back
I thought at first that the pin could not go in properly but in fact it came out as if we doing a pin out: "M280 P0 S10"
which really leaves me thinking about a software bug indeed.

UtterlyD, 2018-11-02:
I also have issues with my genuine BLtouch dropping when probing randomly for a G29. My board is also an MKS SBASE. I have increased Z_CLEARANCE_BETWEEN_PROBES and still get drops just after probe stowing. In addition the probe often does not drop at all causing head crashing.

I can confirm that my previous build from 22 March 2018 does not exhibit this issue, however since updating to bugfix-2.0.x 2 November 2018 @ 1900Hrs I observe this issue.

I have also observed this issue during a G34 command, I have only successfully completed a G34 with 3 cycles once.

Stephane-80, 2018-11-02:
@UtterlyD > In addition the probe often does not drop at all causing head crashing.
I confirm that I also have this problem randomly

p3p, 2018-11-02:
Can anyone confirm the issue is not in https://github.com/MarlinFirmware/Marlin/tree/213e94bce2fd74b10014557bd6801b45193a08f8 but does happen with https://github.com/MarlinFirmware/Marlin/tree/3d5be2ea4bac370e05b8acc61a9fba2684e5131e and tell me the pin used for the servo.

I've tested what I can without a BLTouch, the servo code is outputting the correct waveform for the requested angles on both the hardware and software pins.

manutenfruits, 2018-11-03:
@UtterlyD > In addition the probe often does not drop at all causing head crashing.

Omg what a relief, I wasn't crazy, my wiring was fine! Lol

UtterlyD, 2018-11-03:
@p3p I can confirm that the issue was not present in 
https://github.com/MarlinFirmware/Marlin/tree/213e94bce2fd74b10014557bd6801b45193a08f8 but does indeed occur in 
https://github.com/MarlinFirmware/Marlin/tree/3d5be2ea4bac370e05b8acc61a9fba2684e5131e.

I ran three 4x4 G29s with no premature drops or crashes with the first link, and premature dropped and head crashed on the first G29 row on the second link.

I believe I am using pin 1.23 for the trigger pin. Is there any testing you would like to perform with my logic analyser? Would be a good excuse to find where I put it!

Stephane-80, 2018-11-03:
I am using pin 1.23 too
I have to find the time to test the old version again

p3p, 2018-11-03:
Thanks for that @UtterlyD will help a lot.

p3p, 2018-11-03:
Have you all updated platformio platforms and libraries recently and have `framework-arduino-lpc176x` version `0.0.5`?, I assume you have as it's was pushed a while ago but I thought I better check.

UtterlyD, 2018-11-04:
@p3p I am all up to date including framework-arduino-lpc176x at 0.0.5. I actually setup from scratch on a new MacBook Pro a few days ago.

UtterlyD, 2018-11-04:
@p3p I can confirm that software pin operation does not have this issue. I managed to perform 3x 5x5 G29 without a drop or crash on pin 4.28.

It is probably worth noting that my E0 is on P2_03 which is shared by P1_23.

thinkyhead, 2018-11-04:
@p3p — With the change in servo code did we sacrifice any of our tuning and control parameters, such as `DEACTIVATE_SERVOS_AFTER_MOVE`, `SERVO_DELAY`, `(MIN|MAX)_PULSE_WIDTH`, etc.?

Stephane-80, 2018-11-04:
hello , 
in pins_MKS_SBASE I have :
//
// Servo pin
//
#define SERVO0_PIN         P1_23   // J8-3 (low jitter)
#define SERVO1_PIN         P2_12   // J8-4
#define SERVO2_PIN         P2_11   // J8-5
#define SERVO3_PIN         P4_28   // J8-6

if I change 
Configuration.h with for using SERVO3_PIN for BLTouch
#define Z_PROBE_SERVO_NR 3 

When I compile I have the error :
*** [.pioenvs\LPC1768\src\src\module\configuration_store.cpp.o] Error 1

p3p, 2018-11-04:
If you just want to test on a software PWM pin then change `SERVO0_PIN` to one of the other values. can't reproduce your error with the information you have supplied, I needed to add another 3 values to `SERVO_DELAY` to make it build but that's all.

Stephane-80, 2018-11-04:
hello , thanks 
yes even by configuring SERVO_DELAY I have the error at compilation
`/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
#define NUM_SERVOS 4 // Servo index starts with 0 for M280 command

// Delay (in milliseconds) before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300, 300, 300, 300 }`
tested with SERVO1 - 2 and 3 I have the compilation error.
it does not matter.
at first I wanted to change the servo number to make it clearer in the config file but I ended up with the option to change the pins in effect .

p3p, 2018-11-04:
Can you supply the your complete config files and the pins file (in a zip) you have this compile error with.

p3p, 2018-11-04:
@Stephane-80 thanks, with the supplied configs I can reproduce the error, 

@thinkyhead the generated C++ code does not seem to be valid? I'm not aware of a c++ compiler that supports out of order designated initialisation, `array[num] = {[index] = value };` is C syntax, not standard C++, .. although I may be reading things wrong. I think gcc has extensions that are allowing this to mostly work but I would have to look into it more, and am working on the broken PWM atm.

Stephane-80, 2018-11-04:
@p3p ok, thank you very much, at least it lets me know that it is not my software that bug.

gloomyandy, 2018-11-04:
I had the WProgram.h issue earlier today. I deleted one of the PlatformIO cache directoies (.piolibdeps I think) and rebuilt everything and the problem seems to have gone away, don't you just love PlatformIO!

thinkyhead, 2018-11-04:
@p3p — Yes, I know. I patched that first.

Stephane-80, 2018-11-05:
Hello , thank you very much !
Complilation : OK 
"#define Z_PROBE_SERVO_NR 3" : OK
BLTouch with the pin 4.28 : OK
👍

p3p, 2018-11-07:
@thinkyhead No the hardware PWM pin is actually the best choice there is just something weird going on, I'm tracking down the problem, it's my fault I just don't know why yet.

psavva, 2018-11-11:
I have replicated the same issue.
Let me know if you need any details from me.  For now, +1 on this bug report.
I'm using Re-Arm and genuine BLTouch.

p3p, 2018-11-11:
Thanks @psavva I can reproduce the issue now, I'm still trying to figure out why changing the pulse width of 2 channels in the same period causes it though, it's caused by the default FAN_PIN it is a Hardware PWM pin that gets updated in safe_delay() .. should be fine.. but depending on timing cancels the update for the other hardware channels.

If I can't fix it today I'l roll out a update disabling the hardware PWM until I can fix it.

nemphys, 2018-11-12:
Nice, it also happens to my BLTouch clone.
I am wondering if it also affects the actual probe values, since I am getting very inconsistent results between probes.
Can somebody confirms that probe values are actually *not* affected (as stated above)?
I recently installed the probe so I have no earlier reference. Either it is a bad unit or I should wait to see if the fix changes anything.

nemphys, 2018-11-12:
I have a Anycubic I3 Mega with the Anycubic Trigorilla 1.4 board. I have no idea if it has a LPC176X processor, but the behaviour of the BLTouch is *exactly* as described above (and it only appeared recently).

gloomyandy, 2018-11-12:
@nemphys your board is using an Arduino Mega 2560, so no it does not use an LPC176X. It is very unlikely that the solution being discussed here will help you. There is a small chance it is the same problem and that the issue has not been identified correctly, but that does not seem very likely. So you may want to investigate your issue more deeply and consider posting a new issue.

nemphys, 2018-11-12:
I have no problem posting a new issue, but it would be *identical* to this one (since the title is not processor-specific).
Are you sure you are looking at the right place? Maybe it is caused by something else.

gloomyandy, 2018-11-12:
I'm not the one working on this problem, but the current focus is on an area that will not fix your issue.

Perhaps you should post the details of your system (including configuration files) and the details of what previous version of the firmware worked properly for you and exactly what version does not.

nemphys, 2018-11-12:
OK, let's wait and see if any fixes produced here do anything good in my case and, if not, I will post a new issue.

hidara2000, 2018-11-21:
I'm wondering if this is a problem with the MKS SBASE? I had the same issue with both genuine and non bltouch on smoothieware. Flashed Marlin in the hope that it would disappear. Whilst I still get the odd double drop with Marlin it has improved

p3p, 2018-11-21:
> I'm wondering if this is a problem with the MKS SBASE?

I updated the lpc176x framework yesterday (0.0.6) to disable hardware PWM while I investigate further, this should work around any issues with bltouch.

KiberIrbis, 2018-11-27:
I met with such a problem on my MKS Sbase board in the Chinese clone BLToch. The problem was completely solved only by changing the code - probe.cpp. 
Adding pauses between teams.

```
 bool set_bltouch_deployed(const bool deploy) {
    if (deploy && TEST_BLTOUCH()) {      // If BL-Touch says it's triggered
      bltouch_command(BLTOUCH_RESET);    //  try to reset it.
      safe_delay(20);
      bltouch_command(BLTOUCH_DEPLOY);   // Also needs to deploy and stow to
      safe_delay(20);
      bltouch_command(BLTOUCH_STOW);     //  clear the triggered condition.
      safe_delay(1500);                  // Wait for internal self-test to complete.
                                         //  (Measured completion time was 0.65 seconds
                                         //   after reset, deploy, and stow sequence)
      if (TEST_BLTOUCH()) {              // If it still claims to be triggered...
        SERIAL_ERROR_START();
        SERIAL_ERRORLNPGM(MSG_STOP_BLTOUCH);
        stop();                          // punt!
        return true;
      }
    }

   safe_delay(20);
   (deploy ? BLTOUCH_DEPLOY : BLTOUCH_STOW);

    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (DEBUGGING(LEVELING)) {
        SERIAL_ECHOPAIR("set_bltouch_deployed(", deploy);
        SERIAL_CHAR(')');
        SERIAL_EOL();
      }
    #endif

    return false;
  }
```

hamster65, 2018-12-02:
My bltouch clone just dropped its pin and went into alarm state during print. This happend when I saved the configuration to eeprom (using the graphical reprap discount lcd) after I finetuned the z probe offset during the first layer of the print.

stefan85, 2018-12-05:
> I met with such a problem on my MKS Sbase board in the Chinese clone BLToch. The problem was completely solved only by changing the code - probe.cpp.
> Adding pauses between command.
> 
> ```
>  bool set_bltouch_deployed(const bool deploy) {
>     if (deploy && TEST_BLTOUCH()) {      // If BL-Touch says it's triggered
>       bltouch_command(BLTOUCH_RESET);    //  try to reset it.
>       safe_delay(20);
>       bltouch_command(BLTOUCH_DEPLOY);   // Also needs to deploy and stow to
>       safe_delay(20);
>       bltouch_command(BLTOUCH_STOW);     //  clear the triggered condition.
>       safe_delay(1500);                  // Wait for internal self-test to complete.
>                                          //  (Measured completion time was 0.65 seconds
>                                          //   after reset, deploy, and stow sequence)
>       if (TEST_BLTOUCH()) {              // If it still claims to be triggered...
>         SERIAL_ERROR_START();
>         SERIAL_ERRORLNPGM(MSG_STOP_BLTOUCH);
>         stop();                          // punt!
>         return true;
>       }
>     }
> 
>    safe_delay(20);
>    bltouch_command(deploy ? BLTOUCH_DEPLOY : BLTOUCH_STOW);
> 
>     #if ENABLED(DEBUG_LEVELING_FEATURE)
>       if (DEBUGGING(LEVELING)) {
>         SERIAL_ECHOPAIR("set_bltouch_deployed(", deploy);
>         SERIAL_CHAR(')');
>         SERIAL_EOL();
>       }
>     #endif
> 
>     return false;
>   }
> ```

I changed the code as you described and I can confirm this is working. Before I couldn't get my whole bed probed. Sometimes it wouldn't deploy. Now I probed the whole bed 4 times and no issues so far.

BTopbas, 2018-12-17:
I have same issue. I have BLTouch Clone (Trianglelab store in aliexpress). I just updated my CR-10 with marlin 2.0 bugfix and problem happening. 

Here is video: https://www.youtube.com/watch?v=14cpm1zyswc

If you watch video sometimes pin dropping. But g29 works for me. I don't know probing true or wrong. And i have another problem too. When i control machine with control box. When i push menu button on lcd sometimes bltouch deploy and strow quickly. Not a every press. surfing lcd menu and sometimes deploy when i press whell button. I used same bltouch without problem with 1.1.7. I changed only firmware.

kursatu, 2019-02-11:
I think I see this problem, and it seems like a Marlin defect or something that can be fixed at Marlin layer.
I've found a workaround it and also found what triggers the symptom, see item 1 below.
I have an BIQU SKR 1.1 32 bit arm board and  cloned BL touch.
1- If I home after bl-reset, bed level the probe intermittently deploys.   
2- If I issue bl reset via servo commands, after power on , it does not deploy the probe.
Ie. M280 P0 S160
3- If I reset the bl touch via the same command after bed leveling the probe does not deploy.
With these workaround, put into the start gcodes, I can use my 32 bit board and cloned bl-touch w/o any problems.

boelle, 2019-02-21:
@manutenfruits problem solved?

manutenfruits, 2019-02-21:
@boelle it is for me, but I didn't close the issue. I'm waiting for other people that were affected by it to confirm.

boelle, 2019-02-21:
close it anyway, if others had the problem they would have confirmed long time ago

manutenfruits, 2019-02-21:
Okay, thanks everyone, for the next person looking for the solution: Update marlin :)

closed 2019-02-21 by manutenfruits

boelle, 2019-02-21:
always use the latest bugfix 2.0

Raymo99, 2019-10-04:
Folks, sorry to raise this again, but I have this issue, very repeatable.
I'm using a clone BLtouch.
And using latest Bugfix too, any advice for things to change in the code?

Aurawin, 2019-10-05:
> Folks, sorry to raise this again, but I have this issue, very repeatable.
> I'm using a clone BLtouch.
> And using latest Bugfix too, any advice for things to change in the code?

Today for me too.  Just used existing wire harness and replaced my V3 Authentic to e-Bay clone.  Z-Axis Align, Bilinear Probing works flawslessly.  Somewhere during print the pin lowers... ARRRG.

Raymo99, 2019-10-05:
I suspect your issue is a  bit different to the one in this thread.
If your pin 'falls' during a print it may be related to the mechanics of the Bltouch itself, where there isn't enough magnetic force to hold it up. 
If you remove the grubscrew you can extract the probe for inspection. 
The probe consists of an aluminium rod & a small magnet with heatshrink covering both.
The small magnet is attracted to the grubscrew, this 'holds the probe up' during printing.
You can re-magnetize the grub screw with a strong permanent magnet, this should cure your problem. 

The issue in this thread is a firmware commanded deployment of the probe when it shouldn't, it only happens  during a probing homing session. Otherwise it remains up for the duration of the print.

Aurawin, 2019-10-05:
Nope.  I have that problem.  But the pin lowers between each probe as the XY moves to the next spot.  I'm going to increase my distance between probes value to compensate.