MarlinFirmware/Marlin #12305: UBL probing error

jessehanes, 2018-11-02:
### Description

Setting up UBL and performing initial probing

### Steps to Reproduce

1. G28
2. G29 P1

**Expected behavior:** 
 I expect it to probe until finished

**Actual behavior:** 
Printer probes for a few minutes then stops and reports
20:40:53.470: echo:enqueueing "M80"
20:40:57.699: echo:enqueueing "G28"
20:41:18.736: X:145.00 Y:105.00 Z:11.65 E:0.00 Count X:14336 Y:10417 Z:4640
20:42:55.040: X:145.00 Y:105.00 Z:11.65 E:-50.00 Count X:14336 Y:10417 Z:4640
20:42:55.070: Default storage slot 0 selected.
20:42:55.070: Mesh invalidated. Probing mesh.
20:53:34.604: Error:Probing failed (15)

Roxy-3D, 2018-11-02:
You haven't included your Configuration.h files.    It will be very difficult to help you without knowing how you are trying to configure the machine.

But these 'random' type of failures are very often associated with not having enough 'local memory' available.   If you scale back your mesh size from 15 x 15 to 10 x 10 (for example)...  that will free up some memory and you may get better results.

jessehanes, 2018-11-02:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2542590/Configuration.txt)

Roxy-3D, 2018-11-12:
Memory shouldn't be a problem.   You have plenty of 'head room'.

Roxy-3D, 2018-12-20:
I'm guessing you got everything running OK ?    Can this be closed?

boelle, 2019-02-21:
@jessehanes 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.
Before posting a bug report please test with `bugfix-2.0.x` to check if problem is gone

jessehanes, 2019-02-22:
Yes it is now working properly.

Jesse

boelle, 2019-02-22:
@jessehanes maybe click the close button below?

closed 2019-02-22 by Roxy-3D