MarlinFirmware/Marlin #12307: Missing Servo Pins in pins_Fystec_f6.h

Ahrenfelde, 2018-11-02:
### Description

Missing Servo Pins in pins.h file of the Fystec F6 board.

### Steps to Reproduce
-

**Actual behavior:** 
Pins missing

#### Additional Information
Edit the Servo Pins:

// following are original pins for servos
#define SERVO0_PIN       13
#define SERVO1_PIN       11    // if you enable PS_ON_PIN , CONFLICT!
#define SERVO2_PIN       10    // if you enable FILWIDTH_PIN , CONFLICT!
#define SERVO3_PIN       4     // if you enable RGB_LED,it will CONFLICT!

thinkyhead, 2018-11-04:
Why do you suppose PS_ON, FILWIDTH, and RGB_LED_G pins are being assigned to the extra servo pins? Maybe @mroote has some insight.

mroote, 2018-11-04:
Could you share the change you had to make? I don't see any BLTouch pins in the wiki here: https://wiki.fysetc.com/F6_V1.3/#pin-definition

thinkyhead, 2018-11-05:
608be96

closed 2018-11-05 by thinkyhead