MarlinFirmware/Marlin #12309: Pt100 with mks sbase always shows 300c

keshantao, 2018-11-02:
I tried to set #define TEMP_SENSOR_0  to 147 or 110, but nothing changed
the hotend temp is always 300 when the printer is turned on.
(the good thing is I know the temp controlled fan is working, when it sees 300c, it starts right away, lol)

I am connecting the pt100 directly to my board without amplifier, it was working with smoothies board so I assume the hardware still works just fine, maybe I missed something again in configuration?here is the config file.

thanks.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2544097/Configuration.zip)

gloomyandy, 2018-11-02:
Just to clarify... You had this working with smoothieware using the same control board and the same temperature sensor? Have you changed any other hardware when switching to Marlin from smoothieware (for instance are you using a different display)?

How exactly have you connected the senor to your board, what pins are you using?

You have defined the sensor as type 110 did you also try type 147 (both are pt100 sensors but with different pull ups).

keshantao, 2018-11-03:
I don't change any hardware, not a single connection, I just plug the Pt100  xh2.54 2-pin connector to TH2 and follow the instruction found here http://forum.smoothieware.org/pt100 with Un-amplified PT100
It gives about the same temp on screen as my heated bed, and I autopid it and used it for some prints in PLA just fine.

gloomyandy, 2018-11-03:
So when printing PLA what temperature readings did you get when running smoothieware? 

Unfortunately I don't have a pt100 so can't really test anything for you. But I don't see why there should be any sort of issue with using it with the SBase hardware given that you seem to have it working with Smoothieware. One thing I'm not sure about is how well Marlin is configured for the unamplified pt100, perhaps someone using it on another control board could comment?

keshantao, 2018-11-03:
temperature_control.hotend.slope 0.0523809524 
temperature_control.hotend.yintercept -299.6 

I have this two line in smoothies config, it looks like without the yintercept value it's approx. 300c, but I haven't figured out anything from there .

gloomyandy, 2018-11-03:
This thread may help...
https://github.com/MarlinFirmware/Marlin/issues/2912#issuecomment-268478912
the stuff about the pt100 starts about half way down. The thread gives new values for the Marlin table used to handle the device, you may want to try them. Good Luck! Please update this issue with how you get on.

keshantao, 2018-11-03:
sure I will try that out, in the meantime I found the formula that smoothie use for a liner relation
T = slope * ADC + yintersect when the PT100 is connect without an amplifier
wondering if this can be implemented in marlin in some way.

keshantao, 2018-11-03:
I just changed the thermister 147 file to
  PtLine(0,341,0)
  PtLine(50,406,0)
  PtLine(100,469,0)
  PtLine(150,531,0)
  PtLine(200,591,0)
  PtLine(250,650,0)
  PtLine(300,707,0)
  PtLine(350,763,0)
  PtLine(400,818,0)
and I got E1 Err: maxtemp printer  halted
if I use 4k7 pull up, I get temp of 400 instead of 300.....
I think I either change my hot end to the old one or switch back to smoothie
I like the UBL, maybe I just switch back the old hot end for now

gloomyandy, 2018-11-03:
You seem to be specifying 0 for the pull up resistor value, this is almost certainly wrong.The SBase board has a 4.7K pull up resistor so that3rd entry should probably be 4700.

thinkyhead, 2018-11-04:
```
  PtLine(  0,341,4700)
  PtLine( 50,406,4700)
  PtLine(100,469,4700)
  PtLine(150,531,4700)
  PtLine(200,591,4700)
  PtLine(250,650,4700)
  PtLine(300,707,4700)
  PtLine(350,763,4700)
  PtLine(400,818,4700)
```

keshantao, 2018-11-04:
When I add 4k7 pull up, the temp just jump to 400c :-(

thinkyhead, 2018-11-05:
I believe the MKS SBASE has 3.3v logic, so maybe a different set of scaling factors is needed.

AnHardt, 2018-11-05:
Tables 100 and 147 are for raw PT-resistors. We compare resistances. Voltage is canceled out of the equation..

Only if the PT is behind an amplifier (module) we compare voltages (That's still unpatched - i think). In that case a ADC-input with no pullup is required.

boelle, 2019-02-21:
@keshantao problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead