MarlinFirmware/Marlin #12310: On Tronxy X3A error: 'REPRAPWORLD_KEYPAD_MOVE_MENU' was not declared in this scope

Opilson, 2018-11-02:
At example configuration i had this error.

Fast solution - add this to end of ultralcd.h file

#if ENABLED(ADC_KEYPAD)
      #define REPRAPWORLD_KEYPAD_MOVE_HOME  (buttons_reprapworld_keypad & EN_REPRAPWORLD_KEYPAD_F1)
      #define KEYPAD_EN_C                   EN_REPRAPWORLD_KEYPAD_MIDDLE
    #else
      #define REPRAPWORLD_KEYPAD_MOVE_HOME  (buttons_reprapworld_keypad & EN_REPRAPWORLD_KEYPAD_MIDDLE)
      #define KEYPAD_EN_C                   EN_REPRAPWORLD_KEYPAD_F1
    #endif
#define REPRAPWORLD_KEYPAD_MOVE_MENU (buttons_reprapworld_keypad & KEYPAD_EN_C)

Ludy87, 2018-11-02:
Same https://github.com/MarlinFirmware/Marlin/issues/12299
https://github.com/MarlinFirmware/Marlin/issues/12285

Fix is in this PR https://github.com/MarlinFirmware/Marlin/commit/b31a9300b8a717d9b677b52ecb364b5d10b76dbc

Opilson, 2018-11-02:
Thanks for fast reply!

closed 2018-11-04 by thinkyhead