MarlinFirmware/Marlin #12312: bugfix-2.0.x LiquidTWI2

cdedwards, 2018-11-03:
LiquidTWI2 was attempting to compile for my ReArm board with Viki2 board. WProgram.h is missing and this fails. Needed to add a line to the lib ignore statement. the 30aa480 needs to be on it's own line in the platformio.ini for the LPC1768

lib_ignore        = Adafruit NeoPixel
  30aa480

p3p, 2018-11-03:
With the PlatformIO LDF in the state it is, it tries builds libraries that downloaded for other platforms you've built without checking compatibility and ignores the src folders, I'm probably going to have to disable it again, and just manually add the few libraries lpc176x supports.

cdedwards, 2018-11-03:
so make clean is not working either then? Not sure why I would be even attempting this lib as I have a Viki2 and a RR Full Graphic, one on the ReArm and the other on a 2560. niether of which would be using this lib.

p3p, 2018-11-03:
Make clean doesn't remove downloaded libs just the object files, you would need to delete the `.piolibdeps` folder (or the library from it).

262mike, 2018-11-03:
but I can verify the "lib_ignore" works

dot-bob, 2018-11-03:
LDF seems to be working for me if I use the default mode of chain instead of chain+

thinkyhead, 2018-11-03:
@Bob-the-Kuhn — Any special reason we've chosen to use chain+ instead of chain for our builds?

closed 2018-11-03 by thinkyhead