MarlinFirmware/Marlin #12314: Heater PWM interfering with extruder fan output

nemphys, 2018-11-03:
Hi,
I have a Anycubic I3 Mega printer and I have hooked up a 3-LED strip in parallel with my extruder fan (heatsink fan) and I am facing this issue where the LEDs start to flicker when the extruder heater PID kicks in.

They light up normally together with the extruder auto fan and stay fully on until the target temperature is reached. After that, when apparently the PID routine for the extruder starts doing its job, the lights start to flicker and continue doing so until the print finishes.

Isn't the extruder auto fan supposed to stay fully on all the time (EXTRUDER_AUTO_FAN_SPEED is set at 255)? I have tried switching between hard/soft pwm for the fans, but the problem persists.

Is this normal behaviour, or am I doing something wrong?

PS. I am on the bugfix-1.1.x branch. Trying the same setup while running the stock printer firmware (should be some early 1.1.x), there is also some flickering, but it is very mild (almost not noticeable), which makes me think it is more of a software issue.

thinkyhead, 2018-11-03:
It might be a bug in 1.1.x or a pin conflict that we can only speculate about since you haven't provided your configurations. Since 1.1.x is now end-of-life and `bugfix-2.0.x` has received several updates, including LED code updates, you should migrate to `bugfix-2.0.x` and see if the issue is still present. If it still exists, and it's not a problem with your configuration, then we can proceed to investigate whether it's a bug.

nemphys, 2018-11-03:
Thanks for the ultra fast reply. I will try that, but please note that LED code updates should not have anything to do here, since I am not utilising any LED-related code. I and running the LEDs in parallel with the extruder auto fan. I am attaching my Configuration.h and Configuration_adv.h files, please tell me if you need anything more.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2544808/Configuration.zip)

nemphys, 2018-11-03:
Hmm, enabling AUTO_POWER_CONTROL throws compile errors, will have to investigate

nemphys, 2018-11-03:
I can't compile with AUTO_POWER_CONTROL defined, since there is no PS_ON pin defined in my configuration

thinkyhead, 2018-11-03:
I can't get Marlin `bugfix-1.1.x` to build at all with your configurations:
```
In file included from /var/folders/9l/_vm2dfnn1w3dgmjfjgnzc8tc0000gn/T/arduino_build_134837/sketch/MarlinConfig.h:40:0,
                 from /var/folders/9l/_vm2dfnn1w3dgmjfjgnzc8tc0000gn/T/arduino_build_134837/sketch/G26_Mesh_Validation_Tool.cpp:27:
pins.h:815: error: #error "No E stepper plug left for Z2!"
       #error "No E stepper plug left for Z2!"
        ^
In file included from /var/folders/9l/_vm2dfnn1w3dgmjfjgnzc8tc0000gn/T/arduino_build_134837/sketch/MarlinConfig.h:42:0,
                 from /var/folders/9l/_vm2dfnn1w3dgmjfjgnzc8tc0000gn/T/arduino_build_134837/sketch/G26_Mesh_Validation_Tool.cpp:27:
SanityCheck.h:372: error: #error "Z_DUAL_STEPPER_DRIVERS requires Z2 pins (and an extra E plug)."
   #error "Z_DUAL_STEPPER_DRIVERS requires Z2 pins (and an extra E plug)."
    ^
SanityCheck.h:1111: error: #error "USE_CONTROLLER_FAN requires a CONTROLLER_FAN_PIN. Define in Configuration_adv.h."
     #error "USE_CONTROLLER_FAN requires a CONTROLLER_FAN_PIN. Define in Configuration_adv.h."
      ^
exit status 1
#error "No E stepper plug left for Z2!"
```

nemphys, 2018-11-03:
Yes, I am not running the stock branch, I am using derhopps fork for Anycubic I3 Mega:
https://github.com/derhopp/Marlin-with-Anycubic-i3-Mega-TFT.git

thinkyhead, 2018-11-04:
🤷‍♂️

nemphys, 2018-11-05:
Anyway, I just bought a small MOSFET module in order to power the LEDs directly from the PSU and drive them through a spare pin on the RAMPs, but I still believe this issue should be addressed, since it was not so bad in earlier revisions.

nemphys, 2018-11-05:
Being a developer myself (it is the first time I use Marlin though), I think your comment is a little sarcastic when it shouldn't be.
I am not asking for magical solutions and I know that something like this would be hard to debug.
I just thought you were interested in making your software better and deal with what seems to be a regression.
Anyway, I see that you closed the issue already, so there is nothing to discuss.
Thank you for the time you put into this.

thinkyhead, 2018-11-06:
Sorry that I can't take on your problem. Dozens of people are demanding my time, and I have over 200 issues to triage right now. If you're a developer, then please do as much troubleshooting on your own as you can to help reduce the burden. LED strips wired in parallel with extruder fans is not a supported feature and therefore there's no "regression" to speak of. We support LEDs on their own circuit, and those are working very well, so I suggest you just do LEDs in the supported manner.

nemphys, 2018-11-06:
You're right, the MOSFET solution worked great out of the box.
You might consider adding an extruder led option though, I had to use the red temp led and change the temperature lighting logic in order to achieve the same behaviour as before.