MarlinFirmware/Marlin #12315: [bugfix-2.0.x] Mechanical filament runout sensor problem

jokejoke, 2018-11-03:
### Description

Mechanical "Filament runout sensor" does not work (in bugfix-1.1.x worked normally). M119 shows "trigger" if filament presset and "open" if not, but does not lunch FILAMENT_RUNOUT_SCRIPT during printing. M600 works normally, if it sends by Pronterface.

**Expected behavior:**
Changing state of filament sensor from "trigger" to "open" lunch FILAMENT_RUNOUT_SCRIPT.

**Actual behavior:**
State of filament sensor is changing, but does not luch FILAMENT_RUNOUT_SCRIPT.

#### Additional Information
My [Config.zip](https://github.com/MarlinFirmware/Marlin/files/2544834/Config.zip) (Configuration.h and Configuration_adv.h).

thinkyhead, 2018-11-05:
Is the sensor shown to be enabled in the LCD menu?

jokejoke, 2018-11-05:
I will check it this afternoon.

jokejoke, 2018-11-05:
Yes in the LCD -> Configuration -> Runout Sensor is enabled.

thinkyhead, 2018-11-07:
Hmm… Well, I just tested `bugfix-2.0.x` and with a runout sensor attached to the `Z_MAX_PIN` on a RAMPS board and I got a proper detection, and then it ran my configured script (`M117 Runout Detected`). So it definitely works.

Note that Marlin will only run the configured script if an SD print is in progress or if the print job timer is running (started automatically when the extruder starts heating).

jokejoke, 2018-11-07:
Is it necessary printing from SD card for proper operation of filament sensor? Because i use RPi + Octoprint for this printer.

thinkyhead, 2018-11-07:
**_…or if the print job timer is running (started automatically when the extruder starts heating)_**

jokejoke, 2018-11-07:
I saw the "OR", but for sure :) Is there something, what can I test?

thinkyhead, 2018-11-07:
You could poke through the `runout.h` / `runout.cpp` code and add debug output to see if something weird is going on.

jokejoke, 2018-11-07:
Ok I can try it. Thank you for the hint.

gianmarco49, 2019-02-07:
did you find the solution ? On my tarantula the sensor used to work on previous MArlin version. now it does not anymore (MArlin 2.0). The symtoms are similar to yours. M119 reports the sensor changes  but M600 is not starting. I noticed that if  I start without filament then it works fine ( script triggered)  but removing the filament during a print does not trigger the script

BigE2, 2019-02-08:
Another fail here using OctoPrint. I know it has worked when printing from SD, but well, why print from SD when you have OctoPrint?

InsanityAutomation, 2019-02-08:
There was an issue with some of the block conditions when the runout length was too short awhile ago. Have you verified the current builds, and tried setting runout distance to at least 1mm?

BigE2, 2019-02-08:
I'll check your latest build. Runout distance was set to 5mm. Trying to get through another set of prints here before I can start experimenting again.

gianmarco49, 2019-02-13:
At the moment I did not set the runout distance, should I ?; not sure what the default is but in "Cura" I have added this Gcode
; Default start code
{IF_BED}M190 S{BED}
; Activate all used extruder
; Wait for all used extruders to reach temperature
{IF_EXT0}M109 T0 S{TEMP0}
; Disable Heating to get better probe readings
M104 T0 S0
M140 S0 
; Z Probe
G28;
G29;
; restart heating
{IF_BED}M190 S{BED}
{IF_EXT0}M109 T0 S{TEMP0}

Reason was that according to some reported issue , it was better to do the autolevel procedure after  heating the bed and then restart heating to avoid electrical disturbance.
I noticed that if I do **not** insert the filament, M600 is correctly triggered while executing the autolevel procedure ( M600 fails as the hotend temperature has meanwhile dropped) .
After this phase , print job starts but  looks like the runout sensor is not even checked anymore  as I did triger the sensor several times (manually )  without causing any reaction

FanDjango, 2019-02-13:
By the way @gianmarco49 :

Way behind the times, have you seen **#define PROBING_HEATERS_OFF**?

> Reason was that according to some reported issue , it was better to do the autolevel procedure after heating the bed and then restart heating to avoid electrical disturbance.

The following (as confirmed by the probing repeatability test) is sufficient to make probing very much undisturbed by electrical interference and is a good alternative to the gcode approach:

```
/**
 * Enable one or more of the following if probing seems unreliable.
 * Heaters and/or fans can be disabled during probing to minimize electrical
 * noise. A delay can also be added to allow noise and vibration to settle.
 * These options are most useful for the BLTouch probe, but may also improve
 * readings with inductive probes and piezo sensors.
 */
#define PROBING_HEATERS_OFF       // Turn heaters off when probing §§§§
#if ENABLED(PROBING_HEATERS_OFF)
  //turning the following on, causes G28 Z (Home Z-Axis) to wait for bed heating up, §§§§
  //and causes G29 to wait before each probe, making it really slow. §§§§
  //Bed does not cool down too much, so that's ok for G29. §§§§
  //#define WAIT_FOR_BED_HEATER     // Wait for bed to heat back up between probes (to improve accuracy) §§§§
#endif
#define PROBING_FANS_OFF          // Turn fans off when probing §§§§
//#define PROBING_STEPPERS_OFF      // Turn steppers off (unless needed to hold position) when probing
//#define DELAY_BEFORE_PROBING 200  // (ms) To prevent vibrations from triggering piezo sensors §§§§
```

gianmarco49, 2019-02-13:
No, but I will surely check this weekend , Thank you

boelle, 2019-02-21:
@jokejoke 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.
Before posting a bug report please test with `bugfix-2.0.x` to check if problem is gone

costascontis, 2019-03-08:
i have updated to 2.x.x and filament sensor doent work as it worked at 1.x.x.When starts printing after a while it detects no filament and run M600 .My config
`#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_INVERTING true // set to true to invert the logic of the sensor.
  //#define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.

  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  #define FILAMENT_RUNOUT_DISTANCE_MM 10

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    //#define FILAMENT_MOTION_SENSOR
  #endif
#endif`
I used the Z+ Port, and FIL_RUNOUT_PIN in pins_RAMPS.h to 19

FanDjango, 2019-03-08:
Not sure, but I remember there was a change in the logic some time ago in 2.x.x - consider trying
`#define FIL_RUNOUT_INVERTING false`

seangould, 2019-04-17:
I've been having the same issue with a current build of marlin bugfix-2.0.x running on a re-arm / ramps 1.4 combination. Moving the pin assignment for the filament runout sensor to the z_max pin has the mechanical filament runout sensor working, but it never triggers on the default pin as defined in the pins_ramps_rearm.h file.

Not sure if this is strictly a marlin bug, or an oddity of the re-arm / ramps configuration. I know that the servo pins only show around 3v, not the 5v that you'd normally have on them with an arduino / ramps configuration. 

Hope this helps in diagnosing the problem.

relevant section of config.h to re-assign the pin to use the z_max pin with a re-arm board:

`#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_INVERTING false // set to true to invert the logic of the sensor.
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  #define FIL_RUNOUT_PIN P1_28       // ZMAX pin, since servo pin 4 doesn't seem to work
  // #define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.

  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  //#define FILAMENT_RUNOUT_DISTANCE_MM 25

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    //#define FILAMENT_MOTION_SENSOR
  #endif
#endif
`

hinkey247, 2019-06-04:
seangould - Thank you for you comment - works perfectly.

boelle, 2019-07-21:
i will close this one since OP has not reported back and other have reported it as working

closed 2019-07-21 by boelle

athulsnair, 2019-07-29:
I'm on Marlin 1.1.9 Ramps 1.4.

M600 and Nozzle park works. Tried loading and unloading as well.
Filament sensor shows OPEN when switch is pressed and TRIGGERED when switch is open. 

But it doesn't affect print job. I can continue print or start a new print with Filament sensor TRIGGERED. 

[image]

I tried latest version as well.

Also where this `FILAMENT_RUNOUT_DISTANCE_MM ` ?
I searched both configuration files and couldn't find any

Note: When I tried latest version from github, I copy pasted my configuration file. Is that okay
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3442467/Configuration.zip)

athulsnair, 2019-07-31:
It was my fault. I have set my EXTRUDE_MINTEMP as 0 since I use paste extruder sometimes, and when I was testing the run out sensor, I set the filament temperature as 0 in Slic3r to avoid heating time. Looks like with heater OFF, Marlin will ignore the run out sensor state. When I set filament temperature in Slic3r, it works as expected. 

It is working in Marlin 1.1.9 with default pin(#define FIL_RUNOUT_PIN    4)

JamesKR3D, 2020-04-22:
> Hmm… Well, I just tested `bugfix-2.0.x` and with a runout sensor attached to the `Z_MAX_PIN` on a RAMPS board and I got a proper detection, and then it ran my configured script (`M117 Runout [...]

So does this mean this won't work when printing via USB?  I use S3D to print via USB where I am trying to set up the runout sensor.

gianmarco49, 2020-04-22:
never tried to print via USB  , always SD  but , here my findings
Original Marlyn FW , sensor worked perfectly
I have updated it last year and sensor stopped working; I have tried pretty
much every suggestion  I have found on the site to make it work again
without luck.
Last month I have installed the latest fw release , and it magically
started to work again

JamesKR3D, 2020-04-22:
i couldn't get it to work via usb but it seems to work with an sd print.. which is very odd.. and not very useful for my setup.