MarlinFirmware/Marlin #12319: Bed leveling fade height not working

italocjs, 2018-11-03:
Hello, i'm having trouble enabling the Z leveling fade, i tried several things (and versions) but didnt worked.  I'm trying to use this function during the raft printing, so i can improve the printing accuracy on the Z axis.  

Steps to reproduce (marlin 1.1.9)
>Enable #define ENABLE_LEVELING_FADE_HEIGHT
>Send the gcode M420 S1 Z3   (S1 to enable, Z3 = height to disable)
>Start printing.

Expected behavior:
>printer goes home and make the leveling probings
>printing starts with bed leveling working, at Z=0
>bed leveling fades off gradually until Z=3

Actual behavior
>printer goes home and make the leveling probings
>Printing starts with bed leveling not moving the stepper at Z=?  (not zero, nozzle stays in the air)
>Bed leveling fade dont work, since at layer zero, its already off

The same thing happened with marlin V1.1.9, V1.1.x bugfix, V2.0 bugfix.

Machine:
Custom made,
Drivers TMC2130 on X and Y
Drivers A4988 on Z, E0 and E1
CoreXY
Original BLTouch 
No LCD

gloomyandy, 2018-11-04:
If your print gcode contains a G28 that will turn bed levelling off, you need to do a M420 S1 after any G28.

italocjs, 2018-11-04:
> If your print gcode contains a G28 that will turn bed levelling off, you need to do a M420 S1 after any G28.

Thanks @gloomyandy , but im not sure this is the issue, when i do a G28 then a G29 it should remain active right? or should i use another gcode? here follows my scripts

**Starting script**
>M92 X80.40 Y80.40 ; measured compensation for ABS MG94 3DLAB shrinkage
>M84 S0 ; Keep steppers on
>M104 S0 ; Shutdown the nozzle
>G28 ; go home
>G29 ; autoleveling
>M190 S[bed0_temperature] ; Set bed to target temp and wait
>M109 S[extruder0_temperature] ; set nozzle to target temp and wait
>M420 S1 Z3; enable fading at 3mm

**ending script**
>M104 S0 ; turn off nozzle
>M140 S0 ; turn off bed 
>G1 X200 Y200; move nozzle to a safe position
>M84 ; shutdown steppers

gloomyandy, 2018-11-04:
That looks fine as you have the M420 in your start gcode. It was just that in your original post you said you sent an M420 to the printer before you started printing, this may not be doing anything useful.

As to your problem I'm sorry can't really help much more. I've never used bilinear bed levelling so no experience of how it works with fade height etc.  I assume you have things like your ZOffset set correctly?

italocjs, 2018-11-04:
Yep, if i dont use the M420 the printer works perfectly without any adjustments, which leveling method you use?.  i'm up to try the UBL, but this wouldn't be "easy" to deploy for customers, since its quite a pain to setup the ubl

italocjs, 2018-11-07:
Hello guys, anyone have any idea on how to make it work? ive been looking at the code, but i'm not familiarized with the concept of the mesh bed leveling, so i didnt understood much

gloomyandy, 2018-11-07:
A couple of things you could try....

Change your G code so that you do not perform the G28 and G29 until after the bed has come up to temperature. It could be that the bed is warping a little as it heats up.

Change your "M420 S1 Z3" to just "M420 S1" this may help to identify if the fade out option is causing problems, or if it is just the bed levelling. You could also try "M420 S1 Z10" to see if a larger fadeout helps.

I'm also a little confused as to exactly what the problem is. Is it just that print starts with the nozzle too high or are there other problems? Are you sure that you have your probe offsets set correctly? Perhaps you could post a video clip showing the problem?

italocjs, 2018-11-07:
The problem is that the fade is not happening, and it is not allowing the autoleveling to work.

Without fade: (real)
The probe measures 9 points in the bed, then start printing while moving the Z to compensate the printing surface error.

With fade (expected)
The probe measures 9 points in the bed, then start printing while moving the Z to compensate the printing surface error.  the movement will gradually cease up to 3mm

With fade (real)
The probe measures 9 points in the bed, then start printing but NOT moving the Z to compensate, the nozzle sometimes stays too high, sometimes crash on the bed. 

About the heating, i was doing this, but the bltouch works best when the bed is cold (i measured the repeatability)

gloomyandy, 2018-11-07:
So are you saying that if your startup code says "M420 S1" things work but with "M420 S1 Z3" they do not? Did you try "M420 S1 Z10"? 

As to the bed heating well the probe may be more accurate cold, but if the bed shape changes when it gets hot then you will be measuring the wrong thing more accurately!

italocjs, 2018-11-07:
if use M420 S1, or M420 S1 Z3, the leveling stops working, 
It only work if i leave it off.

I also tried resetting my eeprom, reburning firmware, and the last two bugfixes

gloomyandy, 2018-11-07:
Well there is something strange happening here because ( at least as far as I can see) M29 followed by M420 S1 should in theory be exactly the same as M29 on its own. How are you talking to your printer? If you are using pronterface or repeitier you should see some output from the bed levelling system. 

You could try specifying the Z fade height before you probe the bed. What happens if you insert "G420 Z3" (note no S1) before the G29 in your startup gcode, remove any other G420 entries?

closed 2019-02-19 by thinkyhead