MarlinFirmware/Marlin #12321: Ramps with BLTouch, pin doesn't go down when homing Z

Brakow87, 2018-11-04:
I cant get the pin to drop down when homing Z.  It is a standard Ramps 1.4 board and all wiring is correct.  I cant get the pin to drop using the manual commands either.  Below is my configuration.h  PLEASE HELP my printer is dead!!

Link to Configuration.h and Configuration_adv.h
[https://drive.google.com/drive/folders/1E99sQZHHIeK9RTDPtPCP1tDeRZpk0ivj?usp=sharing](url)

labotecno, 2018-11-04:
#define Z_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // set to true to invert the logic of the probe.
you need to put the same like z min probe endstop inverting  so put true !

kAdonis, 2018-11-04:
You wrote you are using a RAMPS 1.4 but you defined:

`  #define MOTHERBOARD BOARD_RAMPS_13_EFB`

try 

`  #define MOTHERBOARD BOARD_RAMPS_14_EFB`

Test BLTouch with:   M280 P0 S120
BLTouch manual: https://www.antclabs.com/manual

labotecno, 2018-11-04:
ok so you have a problem with your conexion check it more ! whithout moving you axle z you can to stow or desploy the pin of the bltouch with lcd menu ! if you dont have it you have to put it 
other thing comented this to help you make homing separately 
#define INDIVIDUAL_AXIS_HOMING_MENU

italocjs, 2018-11-04:
Something similar happened to me a few times, most of them, there was a bad contact between the servo pin on the ramps and the bltouch.  most of the cases the wiring was bad, on one, the pin was inverted (+ on S, - on -, and S on+)  so the bltouch would light up, but not receive commands.

B3D-P, 2018-11-05:
I just updated to the latest bugfix now and also have the situation where the probe will not work, it goes through power on check fine but then it doesn't work on the LCD or Gcode. Bugfix from a few months ago works fine. I will dig some more but thought to look here to see if any one else was seeing the same thing.

thinkyhead, 2018-11-05:
As one extra troubleshooting step, enable `PINS_DEBUGGING` and then you can use `M43` to check that all the pins are set up correctly. The same option exists in Marlin 1.1.9 (and earlier `bugfix-2.0.x`) so you can compare the output from both to make sure nothing has changed.

Brakow87, 2018-11-05:
Thank you for all of your help!!  It is greatly appreciated!!  I have figured out and resolved the issues.  I found that one of the wires was in fact damaged.  Replaced the wire and changed the board type to ramps_14, and everything came to life.  Thanks again to you all!

closed 2018-11-05 by Brakow87

JuergenBachmann, 2018-11-23:
Hi!
I have got the same problem. When i want to home de z axix, then the pin of the bl-touch is deployed and then retracted. But the homeing isn´t working.
it would be nice, if anybody will have a look to my configoration...
Thanks a lot!

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2611313/Configuration.zip)

kAdonis, 2018-11-23:
Did you change anything or is the BLTouch new? is it a clone or an original?
Try: 
`#define BLTOUCH_DELAY 375   // (ms) Enable and increase if needed`
this is the standard setting

thinkyhead, 2018-11-23:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.