MarlinFirmware/Marlin #12325: 1.1.9: level corners and auto home bug

lujji, 2018-11-04:
### Description
'Level Corners' option is only visible after 'Auto Home' is performed, however this option is still visible after a print is finished. Attempting to perform bed-leveling again will result in axes movement beyond end-stops. In my opinion, this option should always be visible and must perform auto-home by default.

### Steps to Reproduce
1. Auto-home axes
2. Level corners
3. Finish a print
4. Level corners -> results in movement beyond end-stops

thinkyhead, 2018-11-04:
Is the current position somehow corrupted at the end of your prints?

lujji, 2018-11-04:
I don't think the current position is corrupt - I believe level corners still thinks that all axes are homed.

thinkyhead, 2018-11-04:
Try this and see if it fixes it:
```diff
void _lcd_level_bed_corners_homing() {
  _lcd_draw_homing();
- if (all_axes_homed()) {
+ if (all_axes_known()) {
    bed_corner = 0;
    lcd_goto_screen(menu_level_bed_corners);
    _lcd_goto_next_corner();
  }
}
```

lujji, 2018-11-04:
I can't find `_lcd_level_bed_corners_homing` anywhere in the 1.1.9 source..

thinkyhead, 2018-11-04:
I see. The issue has been addressed, then, in `bugfix-2.0.x`. And we won't be patching 1.1.x any more, so if you would like the working version you'll need to install `bugfix-2.0.x`.

lujji, 2018-11-04:
Ok, thanks. In case anyone else encounters this issue, here's a quick workaround. Changes are made in `ultralcd.cpp`.

```diff
void lcd_prepare_menu() {
  ...
  #if ENABLED(LEVEL_BED_CORNERS) && DISABLED(LCD_BED_LEVELING)
-   if (all_axes_homed())
      MENU_ITEM(function, MSG_LEVEL_CORNERS, _lcd_level_bed_corners);
  #endif
}
```

```diff
void _lcd_level_bed_corners() {
  defer_return_to_status = true;
  lcd_goto_screen(_lcd_corner_submenu);
  bed_corner = 0;
- _lcd_goto_next_corner();
+ enqueue_and_echo_commands_P(PSTR("G28"));
}
```

closed 2018-11-05 by thinkyhead