MarlinFirmware/Marlin #12330: Y-Off set at different layer hights

baeriwiliwonka, 2018-11-04:
Hi all,

I just updated to Marlin 1.1.9 and saw a Y layer offset issue several times.
At first I tought it was caused by a filament change, but the Y-offset happened one layer before the layer change and two times after the filament change, so that doesn't seem to be the root cause.

Also in a one coloured print I saw this issue, but at a different layer height.

Problem was independet from the source (Octorprint and SD tested).

I also want to exclude mechanical problems because with a very old firmware (1.1.0) I never had y-offset issues. So I guess somethings wrong in the firmware.

I used the preconfigured config for my CR10S.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

baeriwiliwonka, 2018-11-04:
[image]

swilkens, 2018-11-05:
The step from 1.1.0 to 1.1.9 is a big one. However, considering that your printer is a popular one and we aren't seeing a flood of reports from users with similar issues, it seems likely that this is a mechanical problem.

Some things we might check:

- Is there a significant change in the acceleration, jerk or feedrate values for your printer between the firmware versions?

- Have you checked the sliced gcode on websites such as gcode.ws to verify that this is not an issue in the gcode?

- Have you checked belt tension? They tend to stretch with time.

- Have you tried 2.0.x-bugfix?

Typically, these problems are printer related - not firmware related. You might find more printer specific help on:

- Your printer's facebook group

- Your printer's subreddit

- The official Marlin facebook group (https://www.facebook.com/groups/1049718498464482/)

- The Marlin user's group (https://www.facebook.com/groups/3Dtechtalk/)

boelle, 2019-02-21:
@baeriwiliwonka problem solved in latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead