MarlinFirmware/Marlin #12332: Z_STEPPER_AUTO_ALIGN fails to compile on LPC1768 (MKS SBASE 1.3)

jokeman3000, 2018-11-04:
### Description

When enabling Z_STEPPER_AUTO_ALIGN I receive the following error upon compiling:
`In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:41:0,
from Marlin/src/HAL/HAL_LPC1768/HAL.cpp:23:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:1932:3: error: static assertion failed: Z_STEPPER_ALIGN_[XY]POS settings require one element per Z stepper.
static_assert(
^~~~~~~~~~~~~`

### Steps to Reproduce

1. Uncomment #define Z_STEPPER_AUTO_ALIGN
2. Modify X and Y values for Z_Stepper_align to bed size
3. Build

**Expected behavior:** [What you expect to happen]

Succesful compile

**Actual behavior:** [What actually happens]

Error

#### Additional Information
Attached configurations
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2546328/Archive.zip)

gloomyandy, 2018-11-04:
How many Z stepper drivers do you have? You have defined Z_DUAL_STEPPER_DRIVERS, but you are providing three sense points...
``` C
  // Define probe X and Y positions for Z1, Z2 [, Z3]
  #define Z_STEPPER_ALIGN_X { 10, 100, 210 }
  #define Z_STEPPER_ALIGN_Y { 210, 10, 210 }
```
If you only have two drivers then there should only be a pair of co-ords in each of the above (one for each stepper). If you do have three Z steppers you need to define Z_TRIPLE_STEPPER_DRIVERS rather than Z_DUAL_STEPPER_DRIVERS

jokeman3000, 2018-11-04:
@gloomyandy , thanks for the response. I don't know where my head was, I appear to have misunderstood the "Define probe X and Y positions " comment. Looking back, it makes complete sense now.

Successful compile now. Thanks for pointing me in the right direction!

closed 2018-11-04 by jokeman3000

thinkyhead, 2018-11-04:
@gloomyandy — Oddly, no matter how many times I delete the `.pio*` folders the `WProgram.h` error keeps occurring. I'd say that PlatformIO is being persistently stupid here, but LiquidTWI2 is one of the `common.lib_deps` so it's not wrong for PlatformIO to compile it.

thinkyhead, 2018-11-04:
Patched `platformio.ini` to prevent LiquidTWI2 being compiled under the NXP environments.

p3p, 2018-11-04:
`common.lib_deps` isn't used by the LPC176x builds, and with the strict library option no library that isn't specifically tagged as compatible should be built, so I stick with my PlatformIO being stupid diagnosis.