MarlinFirmware/Marlin #12335: [Bugfix-2.0.x] presentation error Z Value

Ludy87, 2018-11-05:
### Description

wrong Z value when activated `MESH_EDIT_MENU`

`MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_MESH_EDIT_Z, &Z_VALUES(xind, yind), -(LCD_PROBE_Z_RANGE) * 0.5, (LCD_PROBE_Z_RANGE) * 0.5, refresh_planner);`

### Steps to Reproduce

1. `#define AUTO_BED_LEVELING_BILINEAR`
2. `#define LCD_BED_LEVELING`
3. `#define MESH_EDIT_MENU`

**Expected behavior:** Display message is `Z Value:     --.*,(`

**Actual behavior:** Display message should `Z Value:     0.000`

thinkyhead, 2018-11-05:
Unable to reproduce.

Ludy87, 2018-11-05:
I try again and it is same Problem

[Marlin_Config.zip](https://github.com/MarlinFirmware/Marlin/files/2548073/Marlin_Config.zip)

thinkyhead, 2018-11-05:
What does `M420 V` say?

Ludy87, 2018-11-05:
SENDING:M420 V
echo:Bed Leveling Off
echo:Fade Height Off

thinkyhead, 2018-11-05:
I was hoping to see the values of your mesh, but apparently there isn't one…

Ludy87, 2018-11-05:
This problem is before Bed leveling after is all right

thinkyhead, 2018-11-05:
I see. The menu should likely be hidden, then, if `leveling_is_valid` returns `false`.

Ludy87, 2018-11-05:
yes it's better

thinkyhead, 2018-11-05:
…because `NAN` isn't going to display anything pretty.
```cpp
void reset_bed_level() {
  #if ENABLED(DEBUG_LEVELING_FEATURE)
    if (DEBUGGING(LEVELING)) SERIAL_ECHOLNPGM("reset_bed_level");
  #endif
  set_bed_leveling_enabled(false);
  #if ENABLED(MESH_BED_LEVELING)
    mbl.reset();
  #elif ENABLED(AUTO_BED_LEVELING_UBL)
    ubl.reset();
  #elif ENABLED(AUTO_BED_LEVELING_BILINEAR)
    bilinear_start[X_AXIS] = bilinear_start[Y_AXIS] =
    bilinear_grid_spacing[X_AXIS] = bilinear_grid_spacing[Y_AXIS] = 0;
    for (uint8_t x = 0; x < GRID_MAX_POINTS_X; x++)
      for (uint8_t y = 0; y < GRID_MAX_POINTS_Y; y++)
        z_values[x][y] = NAN;
  #elif ABL_PLANAR
    planner.bed_level_matrix.set_to_identity();
  #endif
}
```

Ludy87, 2018-11-05:
or default `0`

thinkyhead, 2018-11-05:
It wouldn't help. The bilinear mesh is never valid until a `G29` has been done to establish the mesh bounds and spacing between points. Having it be "not valid" until that point has a certain utility, as it prevents enabling leveling until data has been acquired.

thinkyhead, 2018-11-05:
93d5b945dd56a6632395522f6c049e942e4b0166

closed 2018-11-05 by thinkyhead