MarlinFirmware/Marlin #12337: platformio : missing configuration for anet

aschor, 2018-11-05:
Because I'm new to platformIO, I tried to find without succes what environment to use with my anet A8.

I couldn't find it anywhere, so I tried uploading with the conf "melzi" because I "thought" that it was some sort of clone. It seems It was not a good choice, because I BRICKED my board. 

After a lot of hassle flashing the bootloader (it's badly documented, and there are all sorts of issues like arduinoISP shut down when connected to anet ICSP headers because of insufficient USB power -?- ) , I tried uploading with original arduino ide with "anet" board, and THERE the ide told me the firmware would not fit on the board (too big).

So my point of view is : 
- there ARE example configuration*.h files for anet boards.
- there WAS a configuration in platformio.ini file for "easy" uploading  ; this configuration was REMOVED by [this ](https://github.com/MarlinFirmware/Marlin/commit/1e00ba54331ed15c10824d0a3936ecb85b510559#diff-237285c047dad808f73698c139144e1a)commit thas was supposed to ADD things for tronxy board, not DELETE other things used by other users (!)
- it is not logical to have configs somewhere for a board, and remove the config for the same board elsewhere. Without notifying. Without comment. 
- instead of REMOVING the code and letting users in the dark because there is no documentation for how to flash safely, nor how to unbrick boards, I suggest that comments in the file explain to users what config to use if they are "supposed" to change their configuration/usage, instead of letting them suppose things, try things, break things. Or have documentation or tutorials that tell users what config to use ? I mean : people buy a printer of some brand, not a "clone of" or a "generic sanguino" (what is sanguino?),  so configurations names must match hardwares name (no ?)

thinkyhead, 2018-11-05:
Our enthusiasm for Anet has waned considerably due to their continued violation of our license and refusal to talk to us. If you want to fix the configuration to make the binary smaller, we'll be happy to accept your changes. But it will be hard to find anyone around here who wants to do the work themselves.

Ludy87, 2018-11-05:
flash for Anet over the Arduino IDE

thinkyhead, 2018-11-05:
Yes, that is correct. The "`anet`" environment was removed because now `sanguino_atmega1284p` should be used.

aschor, 2018-11-05:
> 
> 
> > it is not logical to have configs somewhere for a board, and remove the config for the same board elsewhere. Without notifying. Without comment.
> 
> It is logical for alpha software under development to do whatever it needs to do as it develops, and unfortunately none of the volunteer developers has time right now in their busy lives to write a blog or host a weekly podcast touting every change. If you're playing around with Marlin and PlatformIO, you're on the cutting edge, and so … enjoy the adventure and _caveat emptor_.

that's a good point. I hadn't seen 1.1.8 as alpha software. sorry, I didn't mean anything wrong, in the first place opening this issue was for better user experience (no normal user knows that `sanguino_atmega1284p` should be used for anet and probably other boards/brands )

maybe my bricking was unrelated(?) to the melzi configuration used (since it seems to be the same ?). 

Anyway. this was a suggestion. Closing the issue for now, anet users in search of conf for platformio flash will find it above.

thanks @thinkyhead

closed 2018-11-05 by aschor