MarlinFirmware/Marlin #12341: Feature request: Support for external EEPROM om STM32F4

bipsendk, 2018-11-05:
### Description

Make it possible to use external EEPROM on STM32F4 based board (other STM32 board might support it as well).

The Mini STM32F407VET6 have external 24C08 EEPROM attached, and it would be nice to be able to utilize this components instead of emulating an EEPROM in CPU flash (where the CPU flash has greater risk of being overwritten).

The STM32F4 mini board has the EEPROM connected on pins PB6 (SCL) and PB7 (SDA), which should be a I2C port on the CPU.

Unfortunately I am not a good programmer myself, but has located a I2C library (code) for the STM32F4 which might help/ease the implementation - check https://stm32f4-discovery.net/2014/05/library-09-i2c-for-stm32f4xx/

#### Additional Information

The STM32F407VET6 Mini is achievable at a reasonable price - and will be a good candidate for creating a 32-bit based controller board, where the mini board is plugged into another PCB, that will hold the stepper drivers, LCD interface and so on. This is actually a project, that I am starting on - to try to see if a 32 bit controller can be built at a reasonable price.
More info on the STM32F407VET6 mini at https://wiki.stm32duino.com/index.php?title=Vcc-gnd.com_STM32F407VET6_Mini

ktand, 2018-11-05:
This is already supported in the generic HAL_STM32 using the "official" STM
Core. I have my SCL/SDA pins mapped to PB8/PB9 (it's in the board's
variant.h file).

Make sure you:

#define I2C_EEPROM

in your board's pins-file.

bipsendk, 2018-11-05:
Found it - but not quite sure where the I2C start address is specified.

Will try to change my board's config/pins file and recompile - to check the outcome.

mr-miky, 2018-11-08:
I'm also using it on 24LC64 + STM32F405 with STM32_GENERIC. You must remember to also define the memory capacity of the eeprom.

alexxy, 2018-11-08:
Any ideas how to build with platformio and new hal?

mr-miky, 2018-11-08:
@thinkyhead 
It is already implemented

#define I2C_EEPROM
#define E2END 0xFFF // EEPROM end address (4kB)

hasenbanck, 2018-11-09:
@alexxy I don't think that the "official" STM32 Core is supported by PIO yet. The F4 and F7 HAL mainly target the STM32GENERIC Core and will most likely be phased out in the long run (Since we have a unified HAL with the STM32HAL and the "official" STM32 Core is reaching feature parity with STM32GENERIC soon and is much better maintained).

alexxy, 2018-11-09:
@hasenbanck seems like its supported. But not for all boards. It will be good if its possible to build with pio (since its simplyfyes thisngs a lot). I have STM32F406VET (aka bluepad board)

hasenbanck, 2018-11-09:
@alexxy Ah yeah, now I remember: PIO only support the [core of Roger Clark Melbourne](https://github.com/rogerclarkmelbourne/Arduino_STM32) out of the box. I think some people also have [a running integration](https://github.com/platformio/platform-ststm32/issues/135) of a second core called [STM32GENERIC](https://github.com/danieleff/STM32GENERIC). Both of these cores sadly lack either support for some STM32 systems or lack active maintenance. The current most active STM32 core, which we target with the STM32 HAL is the ["official" STM32 core](https://github.com/stm32duino/Arduino_Core_STM32), since it's . "Official" because some employees of ST are working on it in their free/spare time.

There is an issue about supporting the "official" core [here](https://github.com/platformio/platform-ststm32/issues/76). It seems there is active work in that regard currently undergoing, but no set release date yet.

alexxy, 2018-11-09:
Ok. So now its only possible to build it with arduino?

mr-miky, 2018-11-19:
What address do you refer to?

The eeprom i2c has a physical address that is composed of a part of the CONTROL BYTE (fixed) and the status of the addrss pins.

For a common 24lc64 the fixed part of the CB is 0xC0 therefore the address is (0xC0 >> 1) + the number resulting from the combination of the configuration pins A0, A1, A2.

To make it short, 0x50 + 0,1,2, ... 7 Not knowing the configuration of the pins you have to do only 8 tests ...

However, in HAL / Shared / I2cEeprom.cpp it is already defined eeprom_device_address = 0x50; try to change it ...

vivian-ng, 2018-12-06:
@mr-miky Does
eeprom_device_address = 0x50
mean pins A0, A1, A2 are connected to ground for the 24xx series of EEPROM?

The page on [Smart RAMPS](https://reprap.org/wiki/SMART_RAMPS) with the schematic and config seems to suggest this, but I just wish to confirm my reading of it is correct.

mr-miky, 2018-12-07:
Yes, A0 = A1 = A2 = gnd  - >  0x50

Pay attention that it is a 256 kbit or 32 Kbyte
change the #define
#define E2END 0x7FFF // EEPROM end address (32kB)

The page size is double compared to the one I use but it does not seem to me that marlin does delete / write the eprom in block mode.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-05-16 by thinkyhead