MarlinFirmware/Marlin #12342: Issue with compiling 2.0.X, "error: 'R_SENSE' was not declared in this scope" and others involving "sketch\src\module\stepper_indirection.cpp"

DerEchteLanger, 2018-11-05:
Hi! Im currently trying to compile 2.0.X which i downloaded a few hours ago using Arduino 1.9.0 beta. Since im using TMC2130 drivers in my printer i also downloaded the libary "TMCStepper" version 0.2.2.

When trying to compile i get lots of Errors, all of them from "sketch\src\module\stepper_indirection.cpp":
`Arduino: 1.9.0-beta (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\src\module\stepper_indirection.cpp:159:90: error: 'R_SENSE' was not declared in this scope

     #define _TMC2130_DEFINE(ST, L) TMCMarlin<TMC2130Stepper, L> stepper##ST(ST##_CS_PIN, R_SENSE)

                                                                                          ^

sketch\src\module\stepper_indirection.cpp:160:32: note: in expansion of macro '_TMC2130_DEFINE'

     #define TMC2130_DEFINE(ST) _TMC2130_DEFINE(ST, TMC_##ST##_LABEL)

                                ^

sketch\src\module\stepper_indirection.cpp:164:5: note: in expansion of macro 'TMC2130_DEFINE'

     TMC2130_DEFINE(X);

     ^

sketch\src\module\stepper_indirection.cpp:159:90: error: 'R_SENSE' was not declared in this scope

     #define _TMC2130_DEFINE(ST, L) TMCMarlin<TMC2130Stepper, L> stepper##ST(ST##_CS_PIN, R_SENSE)

                                                                                          ^

sketch\src\module\stepper_indirection.cpp:160:32: note: in expansion of macro '_TMC2130_DEFINE'

     #define TMC2130_DEFINE(ST) _TMC2130_DEFINE(ST, TMC_##ST##_LABEL)

                                ^

sketch\src\module\stepper_indirection.cpp:170:5: note: in expansion of macro 'TMC2130_DEFINE'

     TMC2130_DEFINE(Y);

     ^

sketch\src\module\stepper_indirection.cpp:159:90: error: 'R_SENSE' was not declared in this scope

     #define _TMC2130_DEFINE(ST, L) TMCMarlin<TMC2130Stepper, L> stepper##ST(ST##_CS_PIN, R_SENSE)

                                                                                          ^

sketch\src\module\stepper_indirection.cpp:160:32: note: in expansion of macro '_TMC2130_DEFINE'

     #define TMC2130_DEFINE(ST) _TMC2130_DEFINE(ST, TMC_##ST##_LABEL)

                                ^

sketch\src\module\stepper_indirection.cpp:185:5: note: in expansion of macro 'TMC2130_DEFINE'

     TMC2130_DEFINE(E0);

     ^

sketch\src\module\stepper_indirection.cpp: In function 'void tmc_init(TMCMarlin<TMC2130Stepper, AXIS_LETTER, DRIVER_ID>&, uint16_t, uint16_t, uint32_t, float)':

sketch\src\module\stepper_indirection.cpp:214:23: error: 'INTERPOLATE' was not declared in this scope

     chopconf.intpol = INTERPOLATE;

                       ^

sketch\src\module\stepper_indirection.cpp:219:24: error: 'HOLD_MULTIPLIER' was not declared in this scope

     st.rms_current(mA, HOLD_MULTIPLIER);

                        ^

sketch\src\module\stepper_indirection.cpp: In function 'void reset_stepper_drivers()':

sketch\src\module\stepper_indirection.cpp:609:15: error: 'X_CURRENT' was not declared in this scope

     _TMC_INIT(X, planner.settings.axis_steps_per_mm[X_AXIS]);

               ^

sketch\src\module\stepper_indirection.cpp:143:53: note: in definition of macro '_TMC_INIT'

   #define _TMC_INIT(ST, SPMM) tmc_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                     ^

sketch\src\module\stepper_indirection.cpp:609:15: error: 'X_MICROSTEPS' was not declared in this scope

     _TMC_INIT(X, planner.settings.axis_steps_per_mm[X_AXIS]);

               ^

sketch\src\module\stepper_indirection.cpp:143:67: note: in definition of macro '_TMC_INIT'

   #define _TMC_INIT(ST, SPMM) tmc_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                   ^

sketch\src\module\stepper_indirection.cpp:609:15: error: 'X_HYBRID_THRESHOLD' was not declared in this scope

     _TMC_INIT(X, planner.settings.axis_steps_per_mm[X_AXIS]);

               ^

sketch\src\module\stepper_indirection.cpp:143:84: note: in definition of macro '_TMC_INIT'

   #define _TMC_INIT(ST, SPMM) tmc_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                                    ^

sketch\src\module\stepper_indirection.cpp:615:15: error: 'Y_CURRENT' was not declared in this scope

     _TMC_INIT(Y, planner.settings.axis_steps_per_mm[Y_AXIS]);

               ^

sketch\src\module\stepper_indirection.cpp:143:53: note: in definition of macro '_TMC_INIT'

   #define _TMC_INIT(ST, SPMM) tmc_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                     ^

sketch\src\module\stepper_indirection.cpp:615:15: error: 'Y_MICROSTEPS' was not declared in this scope

     _TMC_INIT(Y, planner.settings.axis_steps_per_mm[Y_AXIS]);

               ^

sketch\src\module\stepper_indirection.cpp:143:67: note: in definition of macro '_TMC_INIT'

   #define _TMC_INIT(ST, SPMM) tmc_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                   ^

sketch\src\module\stepper_indirection.cpp:615:15: error: 'Y_HYBRID_THRESHOLD' was not declared in this scope

     _TMC_INIT(Y, planner.settings.axis_steps_per_mm[Y_AXIS]);

               ^

sketch\src\module\stepper_indirection.cpp:143:84: note: in definition of macro '_TMC_INIT'

   #define _TMC_INIT(ST, SPMM) tmc_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                                    ^

sketch\src\module\stepper_indirection.cpp:630:15: error: 'E0_CURRENT' was not declared in this scope

     _TMC_INIT(E0, planner.settings.axis_steps_per_mm[E_AXIS_N(0)]);

               ^

sketch\src\module\stepper_indirection.cpp:143:53: note: in definition of macro '_TMC_INIT'

   #define _TMC_INIT(ST, SPMM) tmc_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                     ^

sketch\src\module\stepper_indirection.cpp:630:15: error: 'E0_MICROSTEPS' was not declared in this scope

     _TMC_INIT(E0, planner.settings.axis_steps_per_mm[E_AXIS_N(0)]);

               ^

sketch\src\module\stepper_indirection.cpp:143:67: note: in definition of macro '_TMC_INIT'

   #define _TMC_INIT(ST, SPMM) tmc_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                   ^

sketch\src\module\stepper_indirection.cpp:630:15: error: 'E0_HYBRID_THRESHOLD' was not declared in this scope

     _TMC_INIT(E0, planner.settings.axis_steps_per_mm[E_AXIS_N(0)]);

               ^

sketch\src\module\stepper_indirection.cpp:143:84: note: in definition of macro '_TMC_INIT'

   #define _TMC_INIT(ST, SPMM) tmc_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                                    ^

exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`
If i try compiling with TMC2130 2.5.0 i get this Error "sketch\src\module/stepper_indirection.h:59:26: fatal error: TMCStepper.h: No such file or directory" so i assume 0.2.2 is the right one...

My first question would be if this is even the right place for this issue or if i should go to https://github.com/teemuatlut/TMCStepper?

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2550337/configs.zip)

swilkens, 2018-11-05:
https://github.com/MarlinFirmware/Marlin/issues/12144

> Recommend you stick with the old code until @teemuatlut has completed TMCStepper.

DerEchteLanger, 2018-11-05:
The old TMC2130Stepper doesnt seem to work with the most recent Marlin 2.0.X, which version am i supposed to use? My old copy of 2.0.X is from May i think...

teemuatlut, 2018-11-05:
You have an issue with your config files.
```diff
-#if HAS_TRINAMIC
+#if ENABLED(HAVE_TMC2130) || ENABLED(HAVE_TMC2208)
```
Revert that.

DerEchteLanger, 2018-11-05:
Thanks, i copied that over from my old config...

thinkyhead, 2018-11-07:
> Since Github screws up the error [formatting]…

GitHub uses [Markdown](https://guides.github.com/features/mastering-markdown/). You should use three backticks ` ``` ` on a line by itself before and after things that you want to appear as plaintext.

teemuatlut, 2019-01-09:
Two years ago?

MiguelCastillo, 2019-01-10:
lolol! gotcha. Ok i thought this was somehow related to `The old TMC2130Stepper doesnt seem to work with the most recent Marlin 2.0.X` which Im currently running into with an einsy rambo board.

boelle, 2019-02-21:
@DerEchteLanger problem solved?

DerEchteLanger, 2019-02-21:
yes

boelle, 2019-02-21:
then click green close button

closed 2019-02-21 by DerEchteLanger