MarlinFirmware/Marlin #12346: TMC2130 probe cancel sensor less homing z ! Cartesian machine

labotecno, 2018-11-06:
Hello every one ! 5 days ago i try to get sensor less axle z working with TMC2130 with my labrap   , and i finish to get it working .
i use marlin 1.1.9 may be after i will use the bugfix 2 
why the #define FIX_MOUNTED_PROBE cancel the sensorless z ? 
if it can be make some change to do not cancel anything ? 
by the way , only TMC calibration #define TMC_Z_CALIBRATION make z axle calibration why ? 
thank you

thinkyhead, 2018-11-06:
> why the `#define FIX_MOUNTED_PROBE` cancel the sensorless z ?

It doesn't. Where did you get that idea?

> only TMC calibration `#define TMC_Z_CALIBRATION` make z axle calibration why ?

You'll have to ask its author, @teemuatlut, who would be delighted to answer all your questions.

teemuatlut, 2018-11-06:
If you have a probe of any kind you should use that for Z axis homing. That's why the check is there.

`TMC_Z_CALIBRATION` is a script that runs the Z axis to its max to align the left and right side. It doesn't calibrate the drivers but the axis rail.

labotecno, 2018-11-06:
It should very nice to use nozle for probe z , like a Delta , this option miss for tmc2130 sensorless z .

moracabanas, 2018-11-09:
> If you have a probe of any kind you should use that for Z axis homing. That's why the check is there.
> 
> `TMC_Z_CALIBRATION` is a script that runs the Z axis to its max to align the left and right side. It doesn't calibrate the drivers but the axis rail.

Does it work if you conected both Z in paralel with the same TMC2130?

labotecno, 2018-11-09:
Yes It s working but it Will better with Two driver each One motors for current propose

labotecno, 2018-11-11:
https://youtu.be/M9USgFEB0iM its working now i just need get the manual probe enable !!!! for research

closed 2018-11-16 by thinkyhead