MarlinFirmware/Marlin #12347: BLTouch clone probe accuracy better with debugging enabled

drbobbob, 2018-11-06:
### Description

I have an anet A6 that I have installed a BLTouch clone probe (Geetech 3dtouch from ebay).  I have been having issues with inconsistent bed leveling performance.  In an attempt to clean them up I have been trying different configurations and seeing how that affects the M48 performance.  In the latest iteration I noticed that I get very good performance (Range as low as 0.007 in one test, normally around 0.05) when I have debugging all turned on (M111 S38) but worse performance (range around 0.2) when I have debugging turned off (M111 S6)

### Steps to Reproduce

1. Home machine (G28)
2. Run Z probe repeatability test (M48 V4), note performance
3. Turn on debug print statements (M111 S38)
4. Run Z probe repeatability test (M48 V4), not performance

**Expected behavior:** Debug print statements should not affect probe repeatability.

**Actual behavior:** Probing is more consistent with debug print statements enabled.

#### Additional Information

My configuration can be found at https://github.com/drbobbob/Marlin/tree/86389063cd378d5fd01f95436d0e179b5e94e70a

Attaching log of probing.
[ProbeLog.txt](https://github.com/MarlinFirmware/Marlin/files/2551450/ProbeLog.txt)

thinkyhead, 2018-11-06:
ANTCLABS claims their probes are accurate down to 0.0001mm with super-low deviation. You should check your probe repeatability with `M48` and experiment with various slow-probing speeds to figure out the most reliable slow-probe speed to use. Slower is always better.

drbobbob, 2018-11-06:
I've been slowing down my Z slow probe speed and it has helped.  I'm currently homing Z at 240 mm/s fast and 30 mm/s slow.  I have tweaked the offsets so that between probes I am minimizing movement so that it doesn't take forever.

What is concerning to me is that changing the M111 setting affects the results of M48.  I have a few theories that I want to try out, but I wanted to post an issue to see if anyone has better ideas.

I am going to try increasing my BLTOUCH_DELAY to see if that has an affect.
I am also going to try removing the 7805 from the motherboard and using the 5V rail from the powersupply directly, as I suspect the 7805 may be browning out the 5V rail when the BLTouch deploys.

thinkyhead, 2018-11-07:
Such side-effects are hard to pin down, and solutions that work for you to reduce EMF or timing factors may not be universally-applicable. But we'll be curious to see what you discover in your investigations.

bruce356, 2018-11-07:
ANTCLABS Bl touch standard deviation is 0.01mm
Anticlabs comment:- But assembled with the low quality 3D printer, we just get the value as 0.001~0.05mm
regards - bruce

RudyBenner, 2018-11-10:
SENDING:M48 V4
M48 Z-Probe Repeatability Test
Positioning the probe...
Bed X: 110.000 Y: 110.000 Z: 0.003
1 of 10: z: -0.002 mean: -0.0025 sigma: 0.000000 min: -0.002 max: -0.002 range: 0.000
2 of 10: z: 0.003 mean: 0.0000 sigma: 0.002500 min: -0.002 max: 0.003 range: 0.005
3 of 10: z: -0.001 mean: -0.0004 sigma: 0.002125 min: -0.002 max: 0.003 range: 0.005
4 of 10: z: -0.004 mean: -0.0012 sigma: 0.002338 min: -0.004 max: 0.003 range: 0.006
5 of 10: z: -0.002 mean: -0.0015 sigma: 0.002151 min: -0.004 max: 0.003 range: 0.006
6 of 10: z: 0.000 mean: -0.0012 sigma: 0.002041 min: -0.004 max: 0.003 range: 0.006
7 of 10: z: 0.000 mean: -0.0011 sigma: 0.001940 min: -0.004 max: 0.003 range: 0.006
8 of 10: z: -0.001 mean: -0.0011 sigma: 0.001815 min: -0.004 max: 0.003 range: 0.006
9 of 10: z: -0.001 mean: -0.0011 sigma: 0.001712 min: -0.004 max: 0.003 range: 0.006
10 of 10: z: -0.002 mean: -0.0012 sigma: 0.001677 min: -0.004 max: 0.003 range: 0.006
Finished!
Mean: -0.001250 Min: -0.004 Max: 0.003 Range: 0.006
Standard Deviation: 0.001677
>>> M111 S38
SENDING:M111 S38
echo:DEBUG:INFO,ERRORS
SENDING:M48 V4
M48 Z-Probe Repeatability Test
Positioning the probe...
Bed X: 110.000 Y: 110.000 Z: 0.006
1 of 10: z: -0.004 mean: -0.0037 sigma: 0.000000 min: -0.004 max: -0.004 range: 0.000
2 of 10: z: -0.004 mean: -0.0037 sigma: 0.000000 min: -0.004 max: -0.004 range: 0.000
3 of 10: z: -0.006 mean: -0.0046 sigma: 0.001179 min: -0.006 max: -0.004 range: 0.003
4 of 10: z: 0.004 mean: -0.0025 sigma: 0.003750 min: -0.006 max: 0.004 range: 0.010
5 of 10: z: -0.004 mean: -0.0027 sigma: 0.003391 min: -0.006 max: 0.004 range: 0.010
6 of 10: z: -0.004 mean: -0.0029 sigma: 0.003118 min: -0.006 max: 0.004 range: 0.010
7 of 10: z: -0.005 mean: -0.0032 sigma: 0.002977 min: -0.006 max: 0.004 range: 0.010
8 of 10: z: -0.006 mean: -0.0036 sigma: 0.002961 min: -0.006 max: 0.004 range: 0.010
9 of 10: z: 0.003 mean: -0.0029 sigma: 0.003385 min: -0.006 max: 0.004 range: 0.010
10 of 10: z: 0.000 mean: -0.0026 sigma: 0.003328 min: -0.006 max: 0.004 range: 0.010
Finished!
Mean: -0.002625 Min: -0.006 Max: 0.004 Range: 0.010
Standard Deviation: 0.003328

drbobbob, 2018-11-10:
I switched my 5V rail to be directly powered by the power supply.  That did not change the repeatability.

I increased my BLTOUCH_DELAY to 1000 and that seems to have made my probing more consistent. I also think that I am fighting some Z movement consistency because after the printer has been running for a while I think that my Z driver is getting warm and unable to overcome the friction of the Z axis.  I am going to try double checking the positioning of my threaded rod in the couplers as well as adding more lubrication to the Z smooth rods.  I'm considering rewiring the Z motors to be in series rather than parallel because I have read that it will provide a more consistent stepping on the mircostepping.

@RudyBenner Your run with the increased debugging didn't actually change anything because you don't have DEBUG_LEVELING_FEATURE enabled.  The additional logging has to be compiled in.  When I was testing I didn't change the firmware, all had the DEBUG_LEVELING_FEATURE enabled.

With all my changes I can occasionally get probes that have results like: 
22:26:24.598 : Mean: 0.007667 Min: -0.002 Max: 0.018 Range: 0.020
22:26:24.603 : Standard Deviation: 0.006737

I occasionally get larger variation like:
22:36:01.050 : M48 Z-Probe Repeatability Test
22:36:01.053 : Positioning the probe...
22:36:35.325 : Bed X: 111.000 Y: 111.000 Z: -0.052
22:36:56.187 : Bed X: 111.000 Y: 111.000 Z: -0.131
22:36:56.194 : 1 of 10: z: -0.131 mean: -0.1308 sigma: 0.000000 min: -0.131 max: -0.131 range: 0.000
22:37:17.000 : Bed X: 111.000 Y: 111.000 Z: -0.216
22:37:17.005 : 2 of 10: z: -0.216 mean: -0.1733 sigma: 0.042500 min: -0.216 max: -0.131 range: 0.085
22:37:37.815 : Bed X: 111.000 Y: 111.000 Z: -0.242
22:37:37.822 : 3 of 10: z: -0.242 mean: -0.1961 sigma: 0.047348 min: -0.242 max: -0.131 range: 0.111
22:37:58.572 : Bed X: 111.000 Y: 111.000 Z: -0.278
22:37:58.578 : 4 of 10: z: -0.278 mean: -0.2167 sigma: 0.054304 min: -0.278 max: -0.131 range: 0.148
22:38:19.309 : Bed X: 111.000 Y: 111.000 Z: -0.283
22:38:19.316 : 5 of 10: z: -0.283 mean: -0.2300 sigma: 0.055410 min: -0.283 max: -0.131 range: 0.153
22:38:40.087 : Bed X: 111.000 Y: 111.000 Z: -0.300
22:38:40.093 : 6 of 10: z: -0.300 mean: -0.2417 sigma: 0.056913 min: -0.300 max: -0.131 range: 0.169
22:39:00.819 : Bed X: 111.000 Y: 111.000 Z: -0.321
22:39:00.826 : 7 of 10: z: -0.321 mean: -0.2530 sigma: 0.059530 min: -0.321 max: -0.131 range: 0.190
22:39:21.571 : Bed X: 111.000 Y: 111.000 Z: -0.319
22:39:21.578 : 8 of 10: z: -0.319 mean: -0.2612 sigma: 0.059833 min: -0.321 max: -0.131 range: 0.190
22:39:42.313 : Bed X: 111.000 Y: 111.000 Z: -0.324
22:39:42.320 : 9 of 10: z: -0.324 mean: -0.2682 sigma: 0.059776 min: -0.324 max: -0.131 range: 0.193
22:40:03.073 : Bed X: 111.000 Y: 111.000 Z: -0.327
22:40:03.082 : 10 of 10: z: -0.327 mean: -0.2742 sigma: 0.059430 min: -0.327 max: -0.131 range: 0.197
22:40:03.584 : Finished!
22:40:03.586 : Mean: -0.274167 Min: -0.327 Max: -0.131 Range: 0.197
22:40:03.590 : Standard Deviation: 0.059430

I suspect that the larger variance is because I am losing Z steps somewhere due to binding.  As I watch the Z through multiple probes it consistently gets more negative which looks like losing steps while moving positive Z.  Tomorrow I will try cleaning up my Z axis movement to see if that has an effect on consistency.

RudyBenner, 2018-11-10:
I did in fact change M111 as per you instructions. Go back, scan my text

drbobbob, 2018-11-10:
I saw the M111 command.  It didn't change anything because you don't have the DEBUG_LEVELING_FEATURE turned on.  If you look at my original logs, when I send M111 S38 I get a response of "echo:DEBUG:INFO,ERRORS,LEVELING".  The log that you posted had a response of "echo:DEBUG:INFO,ERRORS" which implies that the debugging of leveling isn't turned on.  Also, there are many more debug statements in my probing log.  I didn't see any difference in log statements during your probe.

drbobbob, 2018-11-17:
I now think that I have some mechanical issues with my Z axis.  I am going to close this because I don't think there are firmware issues.

closed 2018-11-17 by drbobbob