MarlinFirmware/Marlin #12349: Auto Bed Leveling: compensation along X axis seems inverted

Jorgininho, 2018-11-06:
### Description

I've installed a capacitive sensor on an Anet A6 to perform autoleveling.
Here are my configurations files : 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2553578/Marlin.zip)

So when I do the autolevel via the command G29 I get this result : 
Bilinear Leveling Grid:
      0      1      2
 0 +0.632 +0.147 -0.153
 1 +0.430 -0.003 -0.290
 2 +0.475 +0.077 -0.217

Seems the bed is tilting around Y axis, ok, I suppose ABL should correct it.
I've set up M851 via the command line to get the right position of the nozzle regarding the bed (M851 Z-1.4, and then nozzle is at good distance when G28, then G0 Z0)
From what I see with the result of the G29, the nozzle sha-ould go lower on column 2 and higher on column 0. But after printing a 1rst layer test shape, its seems the ABL is doing the opposite of the expected behavior. The filament on the left is less compressed that the one on the right of the bed, where I expect then to be even..

thought my issue was similar to this [one ]([url](https://github.com/MarlinFirmware/Marlin/issues/8018))
But I'm not sure, 
- the system is right handed, (0,0) is at bottom left of the bed.
- the inverted compensation behavior remains the same between bilinear and linear autobed leveling
- my printer is an anet A6 where the UBL option could not be used due to limited space on the control board (I"m using platformio to burn the firmware on the printer)

Did I miss a point ? what should I check next ?

Thanks

AnHardt, 2018-11-06:
Come on. There are about 100 issues what handled that.

Most of the end with: "Sorry it's been my hardware."

Never heard about[ twisted x-axis](https://github.com/AnHardt/Marlin/issues/84).

thinkyhead, 2018-11-07:
This issue is caused by a twisted X gantry or some other physical factor messing with your probe readings. This issue is most common on CR-10 and similar machines. There are various adjustments you can try, such as adjusting belt tension, adjusting the rollers, and making sure that cables and bowden tubes have adequate slack.

See: #5983, [#7673](https://github.com/MarlinFirmware/Marlin/issues/7673#issuecomment-351979822), #9529, #10526, and so on…

closed 2018-11-07 by thinkyhead