MarlinFirmware/Marlin #12357: how to update nxplpc-arduino-lpc176x ATOM

gauix4, 2018-11-07:
Hello
and so new on the Atome IDE I would like to know how to update nxplpc-arduino-lpc176x
thank you in advance

[image]

p3p, 2018-11-07:
The `nxplpc-arduino-lpc176x` build platform is downloaded directly from git master branch using the zip link (as can be seen in `platformio.ini`), so it shows as detached because PlatformIOs library manager does not control the versioning, 0.0.1 is the newest version and I don't expect to be changing anything in the build system so it will probably stay that way.

The important one to make sure is up-to-date is `framework-arduino-lpc176x` and this is done using the normal PlatformIO update system as you can see.

thinkyhead, 2018-11-07:
Note that PlatformIO is not going to be updating their Atom plugin going forward, and will be focusing all their efforts on their Visual Studio Code extension. So you may want to switch over at some point.

closed 2018-11-07 by thinkyhead