MarlinFirmware/Marlin #12358: MAX_CMD_SIZE does it need to be that big?

grawi-sikorsky, 2018-11-07:
It's not really issue, but some like question. Does ASCII buffer really needs to be for 96 chars? I'm modifyng marlin 1.1.8 for nextion display support but i'm reaching about 88% SRAM usage. So i'm looking for some space to save precious resource ;) For now longest serial command that i found is M205 (about 50 chars):
eg. "M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.60 E5.00". Changing MAX_CMD_SIZE from 96 to 64 could save a lot of memory with BUFSIZE more than 4.

Best regards for all devs and people involved!

thinkyhead, 2018-11-07:
Marlin allocates all the SRAM buffers it needs at compilation time, except for what may be needed for the stack. The stack itself doesn't get too deep because there are only a couple of interrupts and some occasional stack-based string buffers. We haven't checked lately, but I believe that 1K of free SRAM is more than enough free SRAM to keep the stack from colliding with the globals.

Marlin does strip comments from the commands it receives, and the longest commands produced by slicers for typical 3D printing are probably well within 64 bytes. Given the standard `BUFSIZE` of 4, and placing complete faith in the 64-byte maximum command size, you could recover 128 bytes of SRAM for the stack.

But, if you already have 1K or more free, you're probably on the safe side.

AnHardt, 2018-11-07:
It's configurable to be configured to your needs. Try it!
In worst case you will mess up some parts you want to print.

grawisikorsky-h, 2018-11-07:
Thanks for reply! So maybe I was looking in the wrong direction, blaming low memory. I'm strugling with strange problem for 3 days which i cant explain in different way than "stack-heap kaboom". Sometimes when i start SD print, marlin starts to execute start gcode (shown on serial debug mode M111). After G28 is done, motor movements are stopped but on serial i can see that all commands are processed (in high speed becaue motors are not working so commands goes one after another). I cant find source of the problem, it happens sometimes after i define new function in marlin, or without any changes when i just remove (comment out) some other function. Any ideas?

AnHardt, 2018-11-07:
No movement after G28? Hmmm.
With software endstops ON and no or negative 'way' in between the stops that could be possible.

grawisikorsky-h, 2018-11-07:
I tried to execute M100 or M100 D to see how much memory left before and after stall, but everything looks ok (from 650 to 850bytes free). I dont have idea where to start looking. Most strange for me is that even when i remove some functions (which should save some memory) it could fail. Mostly after G28 it stops (commands are still processed by marlin) but sometimes it execute 2-3 commands more (lets say G1 X0 Y0, G1 Z0.2) but after that another mistery - only E axis is working (much faster because no XY movement).

AnHardt, 2018-11-07:
Try `M211 S0` right behind the `G28`.

grawisikorsky-h, 2018-11-08:
Nope.. M211 S0 turns off software endstops but it doesn't change anything. I tried to undefine software endstops in configuration but still same issue. Normally everything is working but when i enable some functions, objects marlin starts to act like i said before.  For example when i define this:
`
#if ENABLED(NEX_ACC_PAGE)
	NexObject accelin		= NexObject(4, 4, "m5");	// setaccelpagePopCallback -> 
	NexObject Awork			= NexObject(23, 22, "va0");
	NexObject Aretr			= NexObject(23, 23, "va1");
	NexObject Atravel		= NexObject(23, 24, "va2");
	NexObject Amaxx			= NexObject(23, 25, "va3");
	NexObject Amaxy			= NexObject(23, 26, "va4");
	NexObject Amaxz			= NexObject(23, 27, "va5");
	NexObject Amaxe			= NexObject(23, 28, "va6");
	NexObject Asend			= NexObject(23, 33, "p12"); 
	NexObject Asave			= NexObject(23, 30, "p10");	// setaccelsavebtnPopCallback -> settings.save();
	NexObject Aload			= NexObject(23, 31, "p11"); // setaccelloadbtnPopCallback	-> settings.load();
#endif
`

thinkyhead, 2018-11-08:
First, it's recommended you move on to 1.1.9 or 2.0.x, in case whatever you're dealing with has already been patched.

When the steppers stop moving, do they also turn off? Can you move the carriage or the bed by hand?

grawisikorsky-h, 2018-11-08:
Hi! Steppers are going to 0,0,0 pos, and they are holding torque. But after it happens i cant move them in any ways - G1 XYZE is not working. G28,G29 works just fine, but after its done i cant move them again. 
For now I think its problem with my nextion implementation because i cleaned up start-gcode and it's working for a while until i refresh screen. Maybe it's component ID conflict related, i dont know. No way to move on to 1.1.9 because all firmware is heeeavy modified and it'll took ages to move changes to newer version. I'll move to 2.0.x, but first i need to finish this quest ;)

grawisikorsky-h, 2018-11-14:
Just informational: all problems was coused by char array overflow. I was trying to put char[30] into char[20] in one function which was obviously bad idea and i didnt noticed ;)

closed 2018-11-16 by thinkyhead