MarlinFirmware/Marlin #12359: [ 2.0.X ] Compilation fails from NO_LCD_MENU

Transvistinator, 2018-11-07:
Error compiling from the 2.0.X branch after configuring for NO_LCD_MENU.

Looks to be identical or similar to issue #10763

--------------------------
Downloaded the latest version of 2.0.X, configured for my printer and tried to compile with NO_LCD_MENU defined, in order to try and free up space for AUTO_BED_LEVELING_UBL

[https://pastebin.com/78Vu8c40](url)

**Expected behavior:**
It compiles 👍 

**Actual behavior:**
It doesn't compile. 👎 

As per #10763 using SLIM_LCD_MENUS works fine, but doesn't free up enough PROGMEM unfortunately.

Thanks for looking at this 

**Additional Information** 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2558565/Marlin.zip)

Ludy87, 2018-11-07:
you are Have a Melzi-Board or a Anet-Board?

gloomyandy, 2018-11-07:
It would probably help if you supplied details of the compilation error.

Transvistinator, 2018-11-07:
> 
> 
> you are Have a Melzi-Board or a Anet-Board?

Melzi 2.0 with a " Zonestar " LCD

Printer is a Tronxy Zonestar P802M if that helps

thinkyhead, 2018-11-08:
I can't reproduce the issue. I used your configurations and the current `bugfix-2.0.x` and it compiled just fine.

swilkens, 2018-11-08:
Similar issue on Ender-3, different compile errors.

Configs:
[no_lcd_menus_compile_error.zip](https://github.com/MarlinFirmware/Marlin/files/2562139/no_lcd_menus_compile_error.zip)

- BOARD_MELZI_CREALITY

- CR10_STOCKDISPLAY

> sketch\src\lcd\dogm\status_screen_DOGM.cpp: In function 'void _draw_centered_temp(int16_t, uint8_t, uint8_t)':
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp:47:40: error: 'itostr3' was not declared in this scope
> 
>    const char * const str = itostr3(temp);
> 
>                                         ^
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp: In function 'void lcd_implementation_status_message(bool)':
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp:161:47: error: 'status_scroll_offset' was not declared in this scope
> 
>        const char *stat = lcd_status_message + status_scroll_offset;
> 
>                                                ^
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp: In function 'void lcd_impl_status_screen_0()':
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp:217:32: error: 'lcd_blink' was not declared in this scope
> 
>    const bool blink = lcd_blink();
> 
>                                 ^
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp:260:20: error: 'fan_speed' was not declared in this scope
> 
>            blink && fan_speed[0] ? status_screen1_bmp :
> 
>                     ^
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp:282:31: error: 'fan_speed' was not declared in this scope
> 
>          const int per = ((int(fan_speed[0]) + 1) * 100) / 256;
> 
>                                ^
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp:285:36: error: 'itostr3' was not declared in this scope
> 
>            lcd_put_u8str(itostr3(per));
> 
>                                     ^
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp:397:77: error: 'ftostr4sign' was not declared in this scope
> 
>      strcpy(xstring, ftostr4sign(LOGICAL_X_POSITION(current_position[X_AXIS])));
> 
>                                                                              ^
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp:399:76: error: 'ftostr52sp' was not declared in this scope
> 
>      strcpy(zstring, ftostr52sp(LOGICAL_Z_POSITION(current_position[Z_AXIS])));
> 
>                                                                             ^
> 
> sketch\src\lcd\dogm\status_screen_DOGM.cpp:458:46: error: 'itostr3' was not declared in this scope
> 
>      lcd_put_u8str(itostr3(feedrate_percentage));
> 
>                                               ^
> 
> exit status 1
> Fout bij het compileren voor board Sanguino

thinkyhead, 2018-11-08:
Working on it…

Ludy87, 2018-11-08:
@thinkyhead 
that would not be better for @swilkens: 
[pins_SANGUINOLOLU_11.zip](https://github.com/MarlinFirmware/Marlin/files/2562984/pins_SANGUINOLOLU_11.zip)

swilkens, 2018-11-08:
Apologies, I should have made a separate issue report

Transvistinator, 2018-11-08:
> 
> 
> Working on it…

Tested the changes you did in your pull request, compiles like a charm.

Thank you <3

thinkyhead, 2018-11-09:
The PR should be applicable for @swilkens  now too. At least, it compiles with the provided configurations.

swilkens, 2018-11-09:
> The PR should be applicable for @swilkens now too. At least, it compiles with the provided configurations.

It does indeed

closed 2018-11-09 by thinkyhead