MarlinFirmware/Marlin #12360: sunon fans not work

userosos, 2018-11-07:
Hi!
I connected SUNON HA40201V4-A99-A like fans for cooling printing models and it work when i enabled 100% PWM. And not work if a set 50% PWM. Can i change PWM frequency?

swilkens, 2018-11-07:
I had a similar issue and found that the SUNON fans do not handle duty cycle'd DC supply very well. You can find the lower limit where it will still spin by gradually increasing the rate until the fan starts spinning - but you will likely find that the fan refuses to spin until the value is quite high.

Some technical discussion on this can be found here:
http://forum.makergear.com/viewtopic.php?t=6310#p35390

This approach of fan control is a remainder from the brushed fan age, DC brushless fans have private controller boards and need a variable DC voltage for effective speed control.

It would be nice if board manufacturers (and perhaps Marlin?) allow for a variable voltage to the part cooling fans in case of brushless DC fans.

Either way, this is not a direct marlin issue.

rmoravcik, 2018-11-08:
You have to buy a new Sunon fan without 'Auto Restart' feature. That feature is causing it.

userosos, 2018-11-08:
@rmoravcik What it mean -  'Auto Restart'?

rmoravcik, 2018-11-08:
Well, a colleague of mine explained me this feature as a kind of protection, that is trying to automatically restart fan if fan is blocked (not spinning). So, if you try to control such fan with PWM, fan is not spinning enough and auto-restart will detect this as blocked state.

I personally replaced my old Sunon without auto-restart with a new one with auto-restart without knowing that too.

userosos, 2018-11-08:
> Well, a colleague of mine explained me this feature as a kind of protection, that is trying to automatically restart fan if fan is blocked (not spinning). So, if you try to control such fan with [...]

Ok! Big thx! Can i fix it use hardware? Capacity for example?

thinkyhead, 2018-11-08:
You could try a capacitor. No guarantee that it will work, but it will definitely smooth out the current.

userosos, 2018-11-08:
@thinkyhead i connected Capacity 450 uF 65 V - it is work very good.

ManuelMcLure, 2018-11-08:
Please be aware that adding a filtering capacitor may damage the fan MOSFET on your controller board, as described in further posts on the above linked forum thread.

closed 2018-11-09 by thinkyhead

userosos, 2018-11-09:
> Please be aware that adding a filtering capacitor may damage the fan MOSFET on your controller board, as described in further posts on the above linked forum thread.

What happend with MOSFET? I test it 5 mins. MOSFET is cold.

ManuelMcLure, 2018-11-09:
Check out the fifth post on page 2 of http://forum.makergear.com/viewtopic.php?t=6310#p35650 for details.

i-smuglov, 2020-02-19:
Solved it by enabling FAST_PWM_FAN instead of FAN_SOFT_PWM in **configuration.h**
```
#define FAST_PWM_FAN
//#define FAN_SOFT_PWM
//#define SOFT_PWM_SCALE 0
```
Some kickstart may also help
**configuration.adv.h**
`#define FAN_KICKSTART_TIME 1000`