MarlinFirmware/Marlin #12361: Extruder Stepper twitches nervously back and forth with MKS1.4 and DRV8825

MarkusIppy, 2018-11-08:
### Description

Extruder Motor Twitching back and fourth instead of rotating with MKS GEN 1.4 and DRV8825 Stepper drivers . All Other Steppers work correctly with DRV8825

### Steps to Reproduce

1. Configure Marlin 1.1.9 to use DRV8825 for all axis including Extruder 
2. Actuate Extruder 
3. Tell Extruder to extrude 100 mm by sending G1 E100 F100 

**Expected behavior:** Expect Extruder Motor to rotate by by the steps that are sent

**Actual behavior:**Extruder motor twitches back and forth but does not rotate

#### Additional Information

Board is MKS GEN 1.4 with E1 configured as Z2  and MKS 3.2 TFT Touch 
All Steppers are NEMA17 ( Standard ANET A8 with Standard Extruder ) 
All DRV8825 are set to VREF 0.69 V 
All Other Steppers behave correctly 

Steps i took to exclude HW related problems 

-removed  extruder Motor from the nozzle to ensure its not a blocked Nozzle or any other mechanical blockage 

-Unplugging the the X Axis motor at the Board and connecting the E0 Extruder Cable to it moves the extruder stepper as expected when i actuate X Axis  ==> Not Cable or Stepper fault 

-Swapping the X axis and E0 Stepper driver arround to ensure its not a faulty driver == > Not a faulty driver  

-Adjusted VREF from 0.5 to 0.9 for E0 with the same result 

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

[marlinconfig.zip](https://github.com/MarlinFirmware/Marlin/files/2560565/marlinconfig.zip)

landodragon141, 2018-11-08:
What happens if you connect the x axis cable to the e0 driver and try to step it?

MarkusIppy, 2018-11-08:
Hi , Sorry for the delay . When i plug the X Motor cable into the E0 port and type G1 E100 F100 It moves perfectly smooth .

thinkyhead, 2018-11-08:
Does `M502`, `M500`, and a reboot help?

thinkyhead, 2018-11-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

MarkusIppy, 2018-11-08:
> Does `M502`, `M500`, and a reboot help?

no still does the same

closed 2018-11-09 by thinkyhead