MarlinFirmware/Marlin #12363: G30 - Single Z-Probe. How it works?

krajowy, 2018-11-08:
Hi all
I am trying to use Single Z-Probe to 'level' the bed. Sadly with no success ):

In my configuration.h:
#define MESH_BED_LEVELING

I'm trying level the bed running a g code file.

```
G28
G30 P0 X50 Y50 Z-99999 ; probe near an adjusting screw
G30 P1 X100 Y50 Z-99999  
G30 P2 X150 Y50 Z-99999 
G30 P3 X50 Y100 Z-99999 
G30 P4 X100 Y100 Z-99999
G30 P5 X150 Y100 Z-99999 
G30 P6 X50 Y150 Z-99999 
G30 P7 X100 Y150 Z-99999 
G30 P8 X150 Y150 Z-99999 
G30 P9 X101 Y101 Z-99999 S3 

; testing
G1 X130 Y150 Z1 F3000
G1 X170 
G1 X130 
G1 X170 
G1 X130 
G1 X170 
G1 X130 
G1 X170 
G1 X130 
G1 X170 
G1 X130 
G1 X170 
G1 X130
```

I run the code, then the printer probing at pointed coordinates but I can't see any Z adjustments..

Can anyone tell me what is wrong please? 

Thanks

TheSFReader, 2018-11-08:
As far as I understand (Mesh) Bed leveling, you need to use the "G29" command to map the bed, not a sequence of G30 (cf. http://marlinfw.org/docs/gcode/G029-mbl.html ) .

TheSFReader, 2018-11-08:
But it seems you have a probe to do the Bed leveliong measures, in which case you probably should use an other kind of Bed leveling, perhaps UBL ? http://marlinfw.org/docs/features/unified_bed_leveling.html

krajowy, 2018-11-08:
I don't want to use UBL method. To change the area of probing or number of probes you have to change and upload firmware. With G30 Single Z Probe is possible [?] to do it using g code commands. It is much better option for me. 

Look, here, please:
https://reprap.org/wiki/G-code#G30:_Single_Z-Probe

There are some examples but I can't force it to work..

Thanks

TheSFReader, 2018-11-08:
Looks like the uses you refer to is related to the reprap firmware, not the marlin one. If you want to have G30 work in the way described on the reprap wiki, you'll need to use the reprap firmware (or one that has a compatible behaviour).

Looking at the marlin code, Marlin's  G30  _doesn't _ change anything on the bedleveling/calibration.

krajowy, 2018-11-08:
[image]

?
I'm confused...

thinkyhead, 2018-11-08:
This isn't what `G30` does. It just does a single probe and prints the Z value. It is not used for bed leveling.

closed 2018-11-08 by thinkyhead

TheSFReader, 2018-11-08:
The G30 command exists in Marlin. It just doesn't work like in RepRapFirmware. The ReprapWiki documentation could (should ?)  be updated to reflect the difference, like it does for G29.

krajowy, 2018-11-08:
Thanks for help guys..
[but it would be nice to have that feature (: ]