MarlinFirmware/Marlin #12364: [bugfix-2.0.x] slower movements comparing to 1.1.9

h3xman, 2018-11-08:
### Description
I installed bugfix-2.0 on my Ender 3 and noticed that some movements are slower comparing to 1.1.9. When first layers of a test cube are printed diagonal movements seem to be slow and "choppy".

I used the same values for acceleration, max speed etc. in 1.1.9 and bugfix-2.0. Things like linear advance and junction deviation are disabled.

swilkens, 2018-11-08:
I use the same printer and don't have these issues.

Can you please provide:

- You Configuration.h and Configuration_adv.h

- Your printing setup (over USB? From SD?)

- A gcode file that shows this issue

Do you happen to be using Simplify3D?

The only change in speeds I can recall is the homing speed, which was reduced to prevent the end stop switches from bottoming out.

https://github.com/MarlinFirmware/Marlin/issues/12074

h3xman, 2018-11-08:
I'm printing from SD card and slicing with Cura. I'll upload the gcode a bit later. The 1.1.9 version that I was running is actually [TH3D](https://www.th3dstudio.com/knowledge-base/th3d-unified-firmware/). I'll check if there is any difference in speed from vanilla Marlin version with the same config options.

Config files:
[Configuration-bugfix-2.0.zip](https://github.com/MarlinFirmware/Marlin/files/2562104/Configuration-bugfix-2.0.zip)

swilkens, 2018-11-08:
Those configuration files do not match with the latest Ender 3 configuration example files available on this repository.

Can you try again with the latest bugfix-2.0.x and the latest example configuration for your printer? I have done so on my printer and I can't replicate this issue.

h3xman, 2018-11-08:
That's definitely an issue with 2.0. I tested 1.1.9 ([release](https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.9)) and bugfix-2.0 ([this commit](https://github.com/MarlinFirmware/Marlin/tree/525ea554a1e53cef7d60dfa8d33664a99bcd4a3f)) without any modifications: in 1.9 I get fluent and smooth movements and 2.0 stutters and pauses at every corner.

Configs and gcode attached below.

[Configuration-1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/2563067/Configuration-1.1.9.zip)
[Configuration-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/2563068/Configuration-bugfix-2.0.x.zip)
[CE3_xyzCalibration_cube.zip](https://github.com/MarlinFirmware/Marlin/files/2563069/CE3_xyzCalibration_cube.zip)

swilkens, 2018-11-08:
I can see in your gcode that it's making changes to the acceleration and jerk between every single diagonal line on the first layer "skin" (solid infill).

It places a diagonal line, modifies accel and jerk, moves over X or Y without extruding, modifies accel and jerk again, places the next diagonal line.

Visualized in gcode.ws, every small green line is a non-extrusion move. I assume that is where you are seeing these issues?
[image]

> ;TYPE:SKIN
> G1 F2400 E16.59272
> G1 F1800 X100.99 Y119.01 E17.19207
> G1 X100.99 Y100.99 E17.79142
> G1 X119.01 Y100.99 E18.39077
> G1 X119.01 Y119.01 E18.99011
> **M204 S5000**
> **M205 X30 Y30**
> G1 F2400 E13.99011
> G0 F3600 X118.829 Y101.293
> **M204 S500**
> **M205 X20 Y20**
> G1 F2400 E18.99011
> G1 F1800 X118.706 Y101.17 E18.9959
> **M204 S5000**
> **M205 X30 Y30**
> G0 F3600 X118.14 Y101.17
> **M204 S500**
> **M205 X20 Y20**
> G1 F1800 X118.829 Y101.858 E19.02828
> **M204 S5000**
> **M205 X30 Y30**
> G0 F3600 X118.829 Y102.424
> **M204 S500**
> **M205 X20 Y20**
> G1 F1800 X117.574 Y101.17 E19.08729

None of my slicers produce this result, maybe it's a cura feature.

swilkens, 2018-11-08:
Looks like it is.

https://github.com/Ultimaker/Cura/issues/2562

nemphys, 2018-11-09:
I definitely seems to be a Cura issue, since the problem goes away after disabling acceleration and jerk control in Cura.
What is really interesting is that this started to happen *only* after upgrading to Marlin 2.0.x on my printer, without changing *anything* in my Cura settings.
Does this mean that the Cura acceleration/jerk settings were ignored by 1.x.x and started to work with 2.0.x?

swilkens, 2018-11-09:
There was a back and forth change in the way the M204 command parameters work in Marlin, looking at the cura forums it seems that the ultimaker branch of marlin was branched off before or around those changes were made. Maybe that's why this behavior is seen between 1.1.9 and 2.0.x - though I believe that this change was made well before 1.1.9 was released... @thinkyhead ?

Either way, it seems strange to me that cura is setting the acceleration and jerk for every single move. They should just set it at the start of the print and be done with it. Many reports about this can be found on their issue tracker, usually locked with "won't fix" or "pull requests welcome".

This appears to be a slicer issue, not a firmware issue.

thinkyhead, 2018-11-09:
Marlin now does a `planner.synchronize` in response to commands like `M204` that modify parameters it needs to use during planning, which could cause problems if modified in the middle of planning moves. So it's recommended to turn off slicer features that would generate such commands in-between movements.

h3xman, 2018-11-12:
Thanks, disabling "jerk control" does seem to fix it - I suppose it is a slicer issue then. Nevertheless, it would be nice having this info somewhere in the readme, since for many people the initial reaction might be "2.0 is broken, I'll have to revert back to 1.9".

closed 2018-11-12 by h3xman

thinkyhead, 2018-11-12:
Maybe the change I made to `M204` and `M205` can be reverted, but I got the impression that jerk control was causing some bad issues for some users and adding synchronization was the quickest way to fix it. With Marlin 2.0.x under rapid and continuous development, and with it still being an alpha, we are not going to be cataloging every change we make to the code at this time. When it's ready for final release we will produce detailed release notes.

swilkens, 2018-11-15:
For what it's worth, it seems that thinkyhead (for now - so this is likely a temporary situation) has removed the `planner.synchronize` after M204 and M205, so I expect that @h3xman will now no longer have this problem with Jerk and Acceleration control ON in combination with bugfix-2.0.x ?

https://github.com/MarlinFirmware/Marlin/commit/da17f8ad37e5f8895f3db0e8f0cc5988556bafee

thinkyhead, 2018-11-16:
Those `synchronize` additions were a hedge against those settings getting altered while segments were still being joined, since the planner does forward and backward passes as each new segment goes into the queue, and uses those values. There wasn't any specific complaint, so as long as there aren't any strange behaviors in the vicinity of those codes I think it's probably safe to leave out the `synchronize`.