MarlinFirmware/Marlin #12365: [bug] Layer Shifts on bugfix-1.1.9

Anycubic, 2018-11-08:
### Description
After upgrading from 1.1.0 to 1.1.9 I have layer shifts on pretty always same heights on both x and y axis. Reducing speed, acc and jerk, raising vref does not show any improvement. I also got this tested with same printer with magnetic effector and fisheye bearing and I have exactly same issue. 1.1.5 have same issue.

### Steps to Reproduce

1. Turn on printer
2. Print 3Dbenchy with "M420 S1" in the GCODE

If I print same object without "M420 S1" (so leveling disabled) everything is fine

**Expected behavior:** Flawless print

**Actual behavior:** Level shifts at always same heights

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2561620/Marlin.zip)

This is my first Bug report, this is not my native language. I tried my best, let me know what additional information you need.

thinkyhead, 2018-11-09:
1.1.x is end-of-life and will not be receiving any more updates. Please test with `bugfix-2.0.x` and see if the issue is still present in that version of the firmware.

Anycubic, 2018-11-09:
ok, I will do that.
Thank you

boelle, 2019-02-21:
@Anycubic problem solved?

Anycubic, 2019-02-22:
@boelle since then I switched to TMC2130 and still had the issue on 2.x. A few days ago I also changed the board with a Re-arm (hoping that more powerful CPU will sort skippings out), so need to test everything again

boelle, 2019-03-06:
@Anycubic did you re-test?

Anycubic, 2019-03-06:
Sorry but lately I'm very busy with my job, hope to have time to re-test in the next few days

boelle, 2019-03-06:
@Anycubic maybe click close below and open new issue if problem is still there?

Anycubic, 2019-03-06:
Ok, done

closed 2019-03-06 by Anycubic