MarlinFirmware/Marlin #12366: [bugfix-2.0.x]  G28 Or G28 X crashes X axis if the end-stop is in a triggerd state when homing

GhostlyCrowd, 2018-11-08:
### Description

### Steps to Reproduce

G28 Or G28 X crashes X axis if the end-stop is in a triggerd state when homing

1. Home the X axis, with G28, for example my print end script does this to move the tool away from the print.

2. Send a G28 - I do this before every print.

**Expected behavior:**  The Xaxis should detect the X endstop is in a triggerd state already and bump the end-stop, then Y then Z should probe.

**Actual behavior:**  The X endstop does not bump it just tries to move the axis continuously into the already triggered endstop untill you send an emergency stop.

#### Additional Information

This is the third endstop I've tried. The endstop always works  as long as it does not start homing while its in a triggered state. This is 100% reproduce-able every time.  

HOWEVER If you shut the printer off while the X endstop is triggered, turn it back on and Home. It works as it should. This only happens after its homed at least once already. 

Below is a video Showing it.

Please ignor the slight skipped steps on the X, Ive reduced the current dramatically to the bare minimum to move the axis so i dont chew up my belts showing this issue. 

The video shows me homing the printer with G28. Then Homing the X axis with G28 X which parks the axis on the endstop leaving it in a triggered state. I then send G28 to show that instead of detecting its triggering and bumping the endstop it drives it into the end stop crashing the axis until I send M112.

https://youtu.be/4oZtjUbzFqA

thinkyhead, 2018-11-08:
Are you able to narrow down where in the code the error lies?

GhostlyCrowd, 2018-11-08:
I wish i could @thinkyhead  I am by no means a coder, I am very much a mooch of Marlin.  I can tell you it happened After September 17th as that was my previous Pull.  I try to move forward with the bugfix every couple of weeks so i can stay current-ish if i have to report a bug.

thinkyhead, 2018-11-08:
Please ZIP your configurations and attach them to a reply, as requested in the issue template.

GhostlyCrowd, 2018-11-08:
[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/2562798/Marlin_config.zip)
Here you are.

GMagician, 2018-11-08:
@thinkyhead 
```c++
void Endstops::enable(const bool onoff) {
  enabled = onoff;

  #if ENABLED(ENDSTOP_INTERRUPTS_FEATURE)
    update();
  #endif
}
```

I thing that this must be done also when no interrupt enabled

GMagician, 2018-11-08:
@GhostlyCrowd try to remove `#if #endif` lines in endstops.cpp fuction `enable`

GhostlyCrowd, 2018-11-08:
@thinkyhead 

that build Does indeed act properly with my end stops

thinkyhead, 2018-11-08:
@GMagician might have the right idea. Try this one: https://github.com/thinkyhead/Marlin/archive/f96be3c3dd36a13d9de3aae0f831bc604a1c2b73.zip

If that works, I'll submit a patch shortly.

thinkyhead, 2018-11-08:
Ok, since the version from Oct 17 worked, the change must have been after that. Please try this one, from October 28: https://github.com/MarlinFirmware/Marlin/archive/5ead0269676b80292a360e5b132204ebecdc40a5.zip

GhostlyCrowd, 2018-11-08:
@thinkyhead  October 28th Is broken, So something between October 17th and October 28th

GMagician, 2018-11-08:
@GhostlyCrowd and @thinkyhead I think my solution doesn't work because of `ENDSTOP_NOISE_THRESHOLD`. Update must be called I think ENDSTOP_NOISE_THRESHOLD+1 times before it really updates endstops value

GMagician, 2018-11-08:
@GhostlyCrowd you may try this to confim my suspect.
Inside `Enable` function write:
```cpp
Update();
endstop_poll_count=1;
Update();
```

this is no a solution but if it works then that's tha way...

GhostlyCrowd, 2018-11-08:
@GMagician  I cannopt get it to compil adding 

the code you suggested to endstop.cpp 

error: 'Update' was not declared in this scope

GMagician, 2018-11-08:
@GhostlyCrowd sorry I mis typed is all lowercase

GhostlyCrowd, 2018-11-08:
@GMagician  That did indeed fix the problem

Is this a dirty work around or is it the fix?

GMagician, 2018-11-08:
That just prove my doubt and works only for configurations with `ENDSTOP_NOISE_THRESHOLD` enabled, this can't ever be called `dirty work around` it's worst than that ;-)

GhostlyCrowd, 2018-11-08:
@GMagician  Ahhh i understand, so whats the underlying issue you suspect?

GhostlyCrowd, 2018-11-08:
makes sense why i didnt notice this until i had to disable the endstop interrupts because of the TMC library conflicting with it for some odd reason.

GMagician, 2018-11-08:
@thinkyhead is it "legal" to replace `update` with something like `forceupdate` that wait until `endstop_poll_count` is 0?
eg:
```cpp
FORCE_INLINE void forceupdate() {
  #if ENDSTOP_NOISE_THRESHOLD
    do { update(); } while (endstop_poll_count);
  #else
    update();
  #endif
}
```
and remove `update` call inside planner? I think that this last call is not needed because of "always" up to date endstops status when they they are enabled (even with interrupts)

GhostlyCrowd, 2018-11-08:
Ok so @GMagician  your work around seemed to work for homing while the end stop is triggered but it has an adverse effect in the fat that now when my print finishes and it tries to park the X axis with G28 X  its not seeing the endstop trigger and its crashing after the print now.

GMagician, 2018-11-08:
@GhostlyCrowd that should not affect G28.

GhostlyCrowd, 2018-11-08:
@GMagician  You're correct I forgot to increase my stepper current after doing all these tests and it was the sound of it losing steps when bumping. Just did another print to verify.

GMagician, 2018-11-08:
@GhostlyCrowd try replacing `update` call in `enable` with code proposed for `forceupdate`. I'm not sure such blocking code can be done without a `safedelay`? call

GhostlyCrowd, 2018-11-08:
@GMagician  I can do this tomorrow morning for you to test. I have left my workshop for the day.

thinkyhead, 2018-11-09:
> is it "legal" to replace update with something like forceupdate that wait until `endstop_poll_count` is 0?

I think this is starting to look like a hack to work around something fundamentally broken by the earlier update. @ejtagle worked hard to get the endstops checking correct, and the recent change didn't take into account the theory of its mechanism. I would recommend you revert back to before the change was made and reconsider what you were trying to accomplish with the recent change.

ejtagle, 2018-11-09:
@thinkyhead ... you are right. We always (sometimes me too!) fail to get the whole picture here. The problem stated exists, but the fix must be done at a higher level - The stepper and the endstops class is not the problem.

How does this thing work ?
1) If ENDSTOPS_INTERRUPTs is disabled:
-You just get polled endstops every 1khz - If there is NOISE FILTER enabled, then you can get a variable delay between switch press and actually taking it into account, otherwise actions are taken inmediately
2) If ENDSTOPS_INTERRUPTs is enabled, and no filtering, then the switchs are read and action is taken inmediately
3) If ENDSTOPS_INTERRUPTs is enabled, and filtering,, then each interrupts triggers periodic polling (1khz) until values settle down. Then polling ends.

What makes things more complex is that polling or updating state is only done when the endstops are enabled.... That is what is usually causing issues, as endstops, unless ALWAYS_ENABLED, will not update on changes

thinkyhead, 2018-11-09:
Right, so… Instead of a simple enabled state, should we make it have a state 2 … where the endstops are being primed, and a state 1 … where the endstops are now being read for use?

ejtagle, 2018-11-09:
We could try to keep just the state always updated, but not the actions.
That should be easy

GMagician, 2018-11-09:
@ejtagle is it legal to havew interrupt and filter? If I wants interrupt is because I want fast response if I have noise no interrupt is logic to me...

GMagician, 2018-11-09:
@thinkyhead and @ejtagle we may also keep endstop always working when `ENDSTOP_NOISE_THRESHOLD` is enabled (this is the only situation where status is delayed) but not copy `live_state` to `validated_live_state` until endstops are enabled

ejtagle, 2018-11-09:
@GMagician : Yes, it is legal to have interrupts with noise filtering. You can think it as a timer. The interrupt is used to start the timer that polls the endstops. And the timer autoshutdowns when the endstops are stable

@thinkyhead : Most of the issues with the endstops are caused, by, let´s call it, edge triggered actions.
So, if updates are disabled (abort_enabled() returning false), then update() does not execute- There is a risk of losing edges (transitions from non triggered-triggered-non triggered) while disabled.

The other thing that sometimes has caused problems is that there is a delay between the endstop being triggered and Marlin considering it as triggered. This small delay can cause race conditions if movements are too fast, or too short.

What is the possible fix for the G28 command ? - Well, i think the best approach is to slightly modify the flow:

When an G28 command is executed, the first step should be to enable endstops, then perform an small wait (100mS) to allow the noise filter to stabilize, and THEN start the homing process. Alternatively, do a 10mS delay, then wait until endstop_poll_count equals 0.

That is probably the best approach. And 100mS of extra delay when homing is innoticeable! :+1:

thinkyhead, 2018-11-09:
> So, if updates are disabled (`abort_enabled()` returning `false`), then `update()` does not execute

This is not so, at least not before the recent change. `Temperature::isr` calls `poll()` and (when `ENDSTOP_INTERRUPTS_FEATURE` is disabled) `poll()` always calls `update()`. And if `ENDSTOP_NOISE_THRESHOLD` is set then `update()` continues forward…

thinkyhead, 2018-11-09:
It seems to me that the testing of `abort_enabled()` at the very start of `update()` should not be there. The collection of the endstop state should be allowed to proceed no matter what. And then below, where it currently only does logic for `ENDSTOP_NOISE_THRESHOLD` it should also have a block to do logic when `ENDSTOP_NOISE_THRESHOLD` is not set. Specifically, just updating the `live_state` and `old_live_state` values.

ejtagle, 2018-11-09:
@thinkyhead : I think you are right. State should always be kept. The actions is what we disable

ejtagle, 2018-11-09:
That should remove the processing delays

thinkyhead, 2018-11-09:
The downside is that it means the Temperature ISR will always have the added overhead of checking all the endstop pins.

ejtagle, 2018-11-09:
No, because if you use the interrupt on change feature, update should ONLY be called when there is a change. 
And if you are not using that feature, you get continuous polling,

I see your point. The other choice would be to keep it as it is right now, but add the delays i suggested everytime endstops are enabled

thinkyhead, 2018-11-09:
The `forceupdate` approach also makes sense… Keep calling `update()` until the polling count (if existing) goes down to 0. Then we know the endstop state is confirmed and recorded.

thinkyhead, 2018-11-09:
> if you are not using that feature, you get continuous polling,

Currently you would only get continuous polling when using the threshold.

ejtagle, 2018-11-09:
Basically yes. There is something strange here... the update() call should be cancelling any movement against endstops...

thinkyhead, 2018-11-09:
Perhaps the `forceupdate` only needs to be called when turning endstops/probe off.

thinkyhead, 2018-11-09:
Well also, the interrupt won't ever be generated if the switch is already pressed when the move starts.

ejtagle, 2018-11-09:
Exactly, but the state should already reflect at that point that the endstop is pressed

thinkyhead, 2018-11-09:
I've been auditing the procedure we currently follow. Maybe if I continue that process it will make more sense. Did you examine the recent change to see how it causes this issue to now manifest, where it didn't before?

thinkyhead, 2018-11-09:
aa92022

ejtagle, 2018-11-09:
This is incorrect, but the previous code is also incorrect.
If we clear live_state, and we have enabled FILTER_THRESHOLD, then the call to update() from the stepper ISR will never cancel the move.

ejtagle, 2018-11-09:
The reason, as you stated (and you are right) is that there will never be an endstop change interrupt, so there is not enough calls to update to reflect the change of endstop state, and the movement will never be aborted

thinkyhead, 2018-11-09:
Should it only clear `hit_state` then, or just not clear any of those states?

ejtagle, 2018-11-09:
The reason of cleaning both, i guess was to allow a "window of time" with hit_state = 0
Because, if you don´t clean the valid_state, as soon as you call update(), hit_state will be set to the triggered endstop

thinkyhead, 2018-11-09:
I see. Well we call `hit_on_purpose` when we really care about clearing the `hit_state`. So I suppose the order in which we call `not_homing` and `hit_on_purpose` is important.

ejtagle, 2018-11-09:
But certainly this is not correct. valid_state and endstop_poll_count should never be touched, because that interfers with the filter delay

thinkyhead, 2018-11-09:
I concur with that part. The only question, then, is should we "fake" an interrupt and start the polling count whenever re-enabling the endstops so it will start the process of validation?

ejtagle, 2018-11-09:
Exactly! ,, The order is pretty important - you must call hit_on_purpose() when you are sure the endstops are NOT triggered - In the right moment of the calibration process

You don´t need to fake an interrupt. Just setting endstop_poll_count to FILTER_THRESHOLD is enough. 

But if you don´t touch live_state, that is not required

ejtagle, 2018-11-09:
This comment
```
  // Still 'enabled'? Then endstops are always on and kept in sync.
  // Otherwise reset 'live's variables to let axes move in both directions. 
```

Means changing could break what that piece of code was supposed to be fixing. I can´t remember writing it myself...

GMagician, 2018-11-09:
Hey hit_on_purpose is not enough...If you work with interrupt enabled and no threshold and end homing over endstops what happens? live state report that axis is over endstops and if you move away from it live_state is never updated so what happens? you can only move in one direction

thinkyhead, 2018-11-09:
So setting `endstop_poll_count` can be done in `enable()` and `enable_globally()` when the `onoff` parameter is `true`. The call to `hit_on_purpose` is best done after ` (but in practice, I think that means shuffling around all the routines that do endstop-enabled moves).

And then, should the removal of the `#if ENABLED(ENDSTOP_INTERRUPTS_FEATURE)` in aa92022 also be reverted, or is that still sensible?

thinkyhead, 2018-11-09:
We used to only abort moves if the steppers were actually moving _towards_ the triggered endstop.

GMagician, 2018-11-09:
> And then, should the removal of the #if ENABLED(ENDSTOP_INTERRUPTS_FEATURE) in aa92022 also be reverted, or is that still sensible?

Please don't revert it...'live' variables need to be reset after homing

@ejtagle the comment is part of my fix...don't forget what I said above. live_state needs to be cleared after home (or better after endstops are disabled) to avoid issues.

GMagician, 2018-11-09:
> So setting endstop_poll_count can be done in 

if you fake variable you lose filtering action

thinkyhead, 2018-11-09:
FFS — How about reverting to the endstop methodology from 1.1.8?

[76 more replies]

closed 2018-11-09 by GhostlyCrowd