MarlinFirmware/Marlin #12369: STM32F4 (STM32F407VET6 aka BluePad) cannot build with pio

alexxy, 2018-11-08:
Seems like its not possible to build current HAL for STM32F4 with pio with default config

```
In file included from Marlin/src/HAL/HAL_STM32F4/HAL.cpp:30:0:
Marlin/src/HAL/HAL_STM32F4/HAL.h:58:4: error: #error "Serial port 0 does not exist"
#error "Serial port 0 does not exist"
^~~~~
In file included from /home/alexxy/.platformio/packages/framework-arduinoststm32/STM32F4/cores/maple/boards.h:131:0,
from /home/alexxy/.platformio/packages/framework-arduinoststm32/STM32F4/cores/maple/wirish.h:40,
from /home/alexxy/.platformio/packages/framework-arduinoststm32/STM32F4/system/libmaple/Arduino.h:31,
from Marlin/src/HAL/HAL_STM32F4/../../inc/../HAL/HAL_STM32F4/HAL.h:37,
from Marlin/src/HAL/HAL_STM32F4/../../inc/MarlinConfig.h:30,
from Marlin/src/HAL/HAL_STM32F4/HAL_Servo_STM32F4.cpp:26:
/home/alexxy/.platformio/packages/framework-arduinoststm32/STM32F4/variants/generic_f407v/generic_f407v.h:40:0: warning: "CYCLES_PER_MICROSECOND" redefined                                                                                                                 
#define CYCLES_PER_MICROSECOND  168

In file included from Marlin/src/HAL/HAL_STM32F4/../../inc/MarlinConfigPre.h:32:0,
from Marlin/src/HAL/HAL_STM32F4/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_STM32F4/HAL_Servo_STM32F4.cpp:26:
Marlin/src/HAL/HAL_STM32F4/../../inc/../core/macros.h:52:0: note: this is the location of the previous definition
#define CYCLES_PER_MICROSECOND (F_CPU / 1000000UL) // 16 or 20 on AVR

In file included from Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/../HAL/HAL_STM32F4/HAL.h:46:0,
from Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/MarlinConfig.h:30,
from Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/eeprom_emul.h:53,
from Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/eeprom_emul.cpp:53:
Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/../HAL/HAL_STM32F4/watchdog_STM32F4.h:24:8: error: 'IWDG_HandleTypeDef' does not name a type                                                                                                                            
extern IWDG_HandleTypeDef hiwdg;
^~~~~~~~~~~~~~~~~~
In file included from Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/../HAL/HAL_STM32F4/HAL.h:48:0,
from Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/MarlinConfig.h:30,
from Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/eeprom_emul.h:53,
from Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/eeprom_emul.cpp:53:
Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h:84:5: error: 'TIM_HandleTypeDef' does not name a type                                                                                                                           
TIM_HandleTypeDef handle;
^~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h: In function 'uint32_t HAL_timer_get_count(uint8_t)':                                                                                                                           
Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h:108:56: error: 'stm32f4_timer_t {aka struct tTimerConfig}' has no member named 'handle'                                                                                         
return __HAL_TIM_GET_COUNTER(&TimerHandle[timer_num].handle);
^~~~~~
Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h:108:10: error: '__HAL_TIM_GET_COUNTER' was not declared in this scope                                                                                                           
return __HAL_TIM_GET_COUNTER(&TimerHandle[timer_num].handle);
^~~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h:108:10: note: suggested alternative: 'HAL_TIMER_RATE'                                                                                                                           
return __HAL_TIM_GET_COUNTER(&TimerHandle[timer_num].handle);
^~~~~~~~~~~~~~~~~~~~~
HAL_TIMER_RATE
Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h: In function 'void HAL_timer_set_compare(uint8_t, uint32_t)':                                                                                                                   
Marlin/src/HAL/HAL_STM32F4/EEPROM_Emul/../../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h:112:52: error: 'stm32f4_timer_t {aka struct tTimerConfig}' has no member named 'handle'                                                                                         
__HAL_TIM_SET_AUTORELOAD(&TimerHandle[timer_num].handle, compare);
^~~~~~
compilation terminated due to -fmax-errors=5.
In file included from Marlin/src/HAL/HAL_STM32F4/HAL.h:46:0,
from Marlin/src/HAL/HAL_STM32F4/HAL.cpp:30:
Marlin/src/HAL/HAL_STM32F4/watchdog_STM32F4.h:24:8: error: 'IWDG_HandleTypeDef' does not name a type
extern IWDG_HandleTypeDef hiwdg;
^~~~~~~~~~~~~~~~~~
*** [.pioenvs/STM32F4/src/src/HAL/HAL_STM32F4/EEPROM_Emul/eeprom_emul.cpp.o] Error 1
In file included from Marlin/src/HAL/HAL_STM32F4/../../inc/../HAL/HAL_STM32F4/HAL.h:46:0,
from Marlin/src/HAL/HAL_STM32F4/../../inc/MarlinConfig.h:30,
from Marlin/src/HAL/HAL_STM32F4/EmulatedEeprom.cpp:27:
Marlin/src/HAL/HAL_STM32F4/../../inc/../HAL/HAL_STM32F4/watchdog_STM32F4.h:24:8: error: 'IWDG_HandleTypeDef' does not name a type
extern IWDG_HandleTypeDef hiwdg;
^~~~~~~~~~~~~~~~~~
In file included from Marlin/src/HAL/HAL_STM32F4/HAL.h:48:0,
from Marlin/src/HAL/HAL_STM32F4/HAL.cpp:30:
Marlin/src/HAL/HAL_STM32F4/HAL_timers_STM32F4.h:84:5: error: 'TIM_HandleTypeDef' does not name a type
TIM_HandleTypeDef handle;
^~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F4/HAL_timers_STM32F4.h: In function 'uint32_t HAL_timer_get_count(uint8_t)':
[... 42 lines cut ...]
from Marlin/src/HAL/HAL_STM32F4/../../inc/MarlinConfig.h:30,
from Marlin/src/HAL/HAL_STM32F4/HAL_Servo_STM32F4.cpp:26:
Marlin/src/HAL/HAL_STM32F4/../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h:84:5: error: 'TIM_HandleTypeDef' does not name a type
TIM_HandleTypeDef handle;
^~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F4/../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h: In function 'uint32_t HAL_timer_get_count(uint8_t)':
Marlin/src/HAL/HAL_STM32F4/../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h:108:56: error: 'stm32f4_timer_t {aka struct tTimerConfig}' has no member named 'handle'                                                                                                        
return __HAL_TIM_GET_COUNTER(&TimerHandle[timer_num].handle);
^~~~~~
Marlin/src/HAL/HAL_STM32F4/../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h:108:10: error: '__HAL_TIM_GET_COUNTER' was not declared in this scope                                                                                                                          
return __HAL_TIM_GET_COUNTER(&TimerHandle[timer_num].handle);
^~~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F4/../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h:108:10: note: suggested alternative: 'HAL_TIMER_RATE'
return __HAL_TIM_GET_COUNTER(&TimerHandle[timer_num].handle);
^~~~~~~~~~~~~~~~~~~~~
HAL_TIMER_RATE
Marlin/src/HAL/HAL_STM32F4/../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h: In function 'void HAL_timer_set_compare(uint8_t, uint32_t)':
Marlin/src/HAL/HAL_STM32F4/../../inc/../HAL/HAL_STM32F4/HAL_timers_STM32F4.h:112:52: error: 'stm32f4_timer_t {aka struct tTimerConfig}' has no member named 'handle'
__HAL_TIM_SET_AUTORELOAD(&TimerHandle[timer_num].handle, compare);
^~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs/STM32F4/src/src/HAL/HAL_STM32F4/HAL_Servo_STM32F4.cpp.o] Error 1

```
PS It may be good idea to add typical configs for each type of HAL to CI

boelle, 2019-02-21:
@alexxy problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead