MarlinFirmware/Marlin #12371: HELP: 1.1.9 Bilinear bed leveling one side higher than the other

jamesbit, 2018-11-08:
Hallo !

I have a Prusa i3 clone printer which is running Marlin v 1.1.9 and configured to use bilinear bed leveling.
After bed levelling I have one side little higher (about 0.70 babysteps) than the other side.
Is there a command ti compensate the bed like the original Prusa Firmware ?

Thank you....

roel8032, 2018-11-09:
If you use a proximity sensor, at every point it should detect the same amount of aluminium. Do not come to close to the iron of the screws.

RudyBenner, 2018-11-10:
If you have 2 drivers for the 2 Z stepper motors, you can get automatic bed levelling. Works very well. Need 2 or more Z steppers.

ikarisan, 2018-11-13:
Maybe this is a problem I was facing very often with my Anet A(M)8. Relative to the nozzle, where did you mount the sensor? On the side of the nozzle or in the back/front?

thinkyhead, 2018-11-14:
Duplicate of #9529

closed 2018-11-14 by thinkyhead