MarlinFirmware/Marlin #12372: Question. is Atom still recommended for compiling marlinbugfix2.x still or are we pulling away from it for VSCode?

Glod76, 2018-11-08:
Question.
Question. is Atom still recommended for compiling marlinbugfix2.x still or are we pulling away from it for  VSCode?

p3p, 2018-11-08:
If you are building Marlin for a platform not supported in Arduino IDE, you need to build with PlatformIO, Atom was PlatformIOs IDE of choice but they have moved to recommend VSCode, not sure why but it could be because Atom was never very stable, VSCode seems to work better with PlatformIO and is just generaly a better IDE.

So to sum that up we use PlatformIO, which IDE is up to you (you don't even need to use an IDE if you are familiar with the command line), but going with PlatformIOs recommendation of VSCode will probably make your life easier.

GMagician, 2018-11-08:
All p3p said about PlatformIO is valid but I use visual studio community edition, it's free for personal use, and it's better than vscode.
With an addon files are properly filtered and organized in folders.

Roxy-3D, 2018-11-08:
If you are building Marlin Bugfix 2.0.0 for an AVR...    Arduino works fine.    It is the 32-bit platforms that need the more sophisticated build environments.

closed 2018-11-09 by thinkyhead