MarlinFirmware/Marlin #12374: Bed levelling not correct

Arakahn, 2018-11-08:
### Description

I configured my printer to run a 3x3 linear leveling before each print (using a BL touch clone with a double probing at each point to reduce errors). The probing goes well but when printing it looks as if it is not using the leveling data as seen in the pictures (one taken from back left corner the other from the front right).

Here is my start gcode (tried also without the M420 S1, also did a M502 to be safe before the print).

G28 ; home all axes
G29
M420 S1
M420 Z10
G0 X0 Y0 F9000 ; Go to front
G0 Z0.15 ; Drop to bed
G92 E0 ; zero the extruded length
G1 X40 E25 F500 ; Extrude 25mm of filament in a 4cm line
G92 E0 ; zero the extruded length
G1 E-1 F500 ; Retract a little
G1 X80 F4000 ; Quickly wipe away from the filament line
G1 Z0.3 ; Raise and begin printing.

Also attaching my configuration.h

[image]
[image]

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2564041/Configuration.zip)

thinkyhead, 2018-11-09:
We're very confident there is no bug in Marlin's handling of mesh leveling. If you're not getting perfect results on one side or another, this points to a mechanical anomaly in your printer, which could be due to any of several factors. Meshes can be modified and saved to EEPROM, so we recommend you adjust the points that seem too high or too low, save your improved mesh to EEPROM, and use `M420 S1` without `G29` at the start of every print.

swilkens, 2018-11-09:
I assume you checked if there was a mesh generated that makes sense?

Some other things that might be a factor here:

- The clones aren't always as reliable as the original in terms of error rate

- Reducing the probing speed might improve accuracy

- Probe at printing bed temperature to make sure you account for bed warping

Ahrenfelde, 2018-11-10:
I got the same issue. Switch first do linear bed leveling, it works better. Now i switch to klipper firmware.

closed 2019-02-19 by thinkyhead