MarlinFirmware/Marlin #12377: Latest Marlin bugfix 2.0.x ultralcd.cpp fails to compile with Zonestar LCD (Anet A8 LCD)

Beherith, 2018-11-09:
I have enabled #define ZONESTAR_LCD as usual for my RAMPS build, which compiled fine about a month ago, but yesterdays version of marlin doesnt compile: 

...\arduino_build_202321\sketch\src\lcd\ultralcd.cpp: In function 'bool handle_adc_keypad()':
...\arduino_build_202321\sketch\src\lcd\ultralcd.cpp:236:72: error: 'menu_action_back' was not declared in this scope

         else if (RRK(EN_REPRAPWORLD_KEYPAD_LEFT))   { menu_action_back();     lcd_quick_feedback(true); }

1. #define ZONESTAR_LCD    
2. Compile
**Expected behavior:** Successful compilation

**Actual behavior:** The error shown above.

#### Additional Information

Configuration and configuration_adv below: 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2565003/Marlin.zip)

Ludy87, 2018-11-09:
Update your version to [Bugifx-2.0.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)

Beherith, 2018-11-09:
This is bugfix-2.0.x, I downloaded the source from github yesterday.

RudyBenner, 2018-11-09:
Ludy : typo in that link. The URL is good.

Arduino: 1.8.7 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\src\lcd\ultralcd.cpp: In function 'bool handle_adc_keypad()':

sketch\src\lcd\ultralcd.cpp:236:72: error: 'menu_action_back' was not declared in this scope

         else if (RRK(EN_REPRAPWORLD_KEYPAD_LEFT))   { menu_action_back();     lcd_quick_feedback(true); }

                                                                        ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Beherith, 2018-11-10:
Fixed, thank you very much!

closed 2018-11-10 by Beherith