MarlinFirmware/Marlin #12381: Syntax for G29 S3 and S4

boelle, 2018-11-09:
i had a look here http://marlinfw.org/docs/gcode/G029-mbl.html

S4 i guess the the syntax is G29 S4+0.45 to lift the whole mesh 0.45mm

But what is the syntax for S3 ?

Roxy-3D, 2018-11-09:
Are you using the original Mesh Based Leveling system?    Because that is what those commands are for....

boelle, 2018-11-09:
manual level in this case

boelle, 2018-11-09:
"With Mesh Bed Leveling (MBL) you can interactively measure a grid of Z heights without a bed probe."

i just need an example on how to use G29 S3

boelle, 2018-11-09:
G29 S3 X125 Y105

that that how it works to correct the point at X125 Y105 ?

Roxy-3D, 2018-11-09:
I haven't used the original Mesh Bed Leveling.  But here is the code for what you are talking about.  It does look for an X & Y parameter:
```

    case MeshSet:
      if (parser.seenval('X')) {
        px = parser.value_int() - 1;
        if (!WITHIN(px, 0, GRID_MAX_POINTS_X - 1)) {
          SERIAL_PROTOCOLPAIR("X out of range (0-", int(GRID_MAX_POINTS_X));
          SERIAL_PROTOCOLLNPGM(")");
          return;
        }
      }
      else {
        SERIAL_CHAR('X'); echo_not_entered();
        return;
      }

      if (parser.seenval('Y')) {
        py = parser.value_int() - 1;
        if (!WITHIN(py, 0, GRID_MAX_POINTS_Y - 1)) {
          SERIAL_PROTOCOLPAIR("Y out of range (0-", int(GRID_MAX_POINTS_Y));
          SERIAL_PROTOCOLLNPGM(")");
          return;
        }
      }
      else {
        SERIAL_CHAR('Y'); echo_not_entered();
        return;
      }
```

closed 2018-11-09 by boelle

Roxy-3D, 2018-11-09:
Oh!   You were giving it a coordinate (in mm).   It wants a mesh index.  0 - (GRID_MAX_POINTS_X-1)

boelle, 2018-11-09:
yeah... its hard to guess how the syntax should be when there is no example at all

ie look at the link in my first post and look at like you dont know how to code and are just a user

boelle, 2018-11-09:
so with 3x3 (9) points say i wanted to adjust the one in the middle

how would i do that? ie if i want to rase that point 0.1mm (its to close to bed)

Roxy-3D, 2018-11-09:
Agreed.    But with that said...  The original mesh bed leveling is mostly being kept around for legacy reasons.    It is so small...  It was decided it should be kept as an option for people that need a very small solution.   And the other mesh based leveling systems can worry less about being bloated.

Roxy-3D, 2018-11-09:
> how would i do that? ie if i want to rase that point 0.1mm (its to close to bed)

That would be M421 I think....    Or...  if you have UBL, you can use the interactive mesh editor.

boelle, 2018-11-09:
M420 S1?? that is just to use the mesh

i guess what you call original mesh bed level is using a sheet of paper?

boelle, 2018-11-09:
but yes it is the MESH_BED_LEVELING option in configuration.h

but oh well... maybe @thinkyhead could add more examples to the documentaion
http://marlinfw.org/docs/gcode/G029-mbl.html

there is one example for S0

Roxy-3D, 2018-11-09:
M421 is to set a mesh point value.    

```
/**
 * M420: Enable/Disable Bed Leveling and/or set the Z fade height.
 *
 *   S[bool]   Turns leveling on or off
 *   Z[height] Sets the Z fade height (0 or none to disable)
 *   V[bool]   Verbose - Print the leveling grid
 *
 * With AUTO_BED_LEVELING_UBL only:
 *
 *   L[index]  Load UBL mesh from index (0 is default)
 *   T[map]    0:Human-readable 1:CSV 2:"LCD" 4:Compact
 *
 * With mesh-based leveling only:
 *
 *   C         Center mesh on the mean of the lowest and highest
 *
 * With MARLIN_DEV_MODE:
 *   S2        Create a simple random mesh and enable
 */
```

thinkyhead, 2018-11-10:
Ok, updated http://marlinfw.org/docs/gcode/G029-mbl.html with an S3 example and added the XYZ parameters, which were left out.