MarlinFirmware/Marlin #12383: [FR] Allow decimal on probe offset XY

boelle, 2018-11-09:
might sound stupid, but why are decimals only allowed for Z direction and not XY ??

Roxy-3D, 2018-11-09:
This is because of the sanity checks that are done on the user's configuration values.   The C pre-processor is pretty lame.  It can only do integer math.   So the X_OFFSET_FROM_EXTRUDER (and Y value) need to be integers to do the comparisons.

boelle, 2018-11-09:
hehe like the arduino ide/compiler dont like comments after a setting
so at least its not a marlin thing, more like the pre-processor is forcing marlin

boelle, 2018-11-09:
reason is that on one of my printers i have one of these fancy mk42 beds and i got the config just so it hits the 9 probe points, the 3 to the left are spot on, but the 3 to the right are off by 1 mm in x direction, but the nozzle itself can hit each probe point excatly in the center. and the only logical for the probe not hitting precise in center would be the offset

its one of those things that my head just cant figure... why is first probe spot on but the 3rd is off. 4 spot on and then 6th is off

thinkyhead, 2018-11-17:
It is possible to use float values with a few small changes to the code. But note that using float values for these offsets does increase the code size by over 200 bytes with a minimal configuration. That said, a difference of less than a millimeter in the probe offsets should have zero discernible effect on probe results.

boelle, 2018-11-17:
the problem comes when you try to hit a 8mm probe spot with an 8mm probe, first probe is spot on, the next a bit off and the 3rd is more etc. etc.

thinkyhead, 2018-11-17:
There should be no cumulative error related to the probe offsets. The offsets are simply added to the position on each move to some probe point, and they are not accumulated over space or time. If you’re seeing an error of that kind, it would be more likely related to dividing up the total space by the number of probe points.

boelle, 2018-11-17:
well i have set it to 3 points on each axis since there are 9 in total and i know where each point are located

funny thing is that i can make the nozzle touch the center of each point, so it's a bit of a puzzle why  the probe cant

but will double check config tomorrow, need a bit of sleep first :-D

thinkyhead, 2018-11-17:
Well, you can confirm or refute the cause by adding 0.5 the probe offsets and seeing what happens.

closed 2018-12-06 by boelle